OpenEPaperLink/OpenEPaperLink #1: Heap size

nlimper, 2023-01-21:
In newproto.cpp: Op dit moment laadt je de hele bmp-file in pendinginfo->data op het moment dat het eerste block aangevraagd wordt door de tag. Een image is 128*296*3 bytes groot, dat is 133kB, en dat past niet in de heap (ESP.getMaxAllocHeap() -> 110580).
De processor crashet, en vervolgens heeft de tag-firmware een oneindige retry, en blijft continu via zigbee vragen om dat block.
Is het niet beter om processBlockRequest() alleen de benodigde chunk van het bestand op te laten halen? Dan hoeven de plaatjes niet in z'n geheel in memory gelezen te worden.

jjwbruijn, 2023-01-21:
Goeie vraag!

Dat is zeker een oplossing, maar introduceert extra latency op het moment dat een block gevraagd wordt. Niet groot, maar LittleFS is ook weer niet briljant snel, en hoe sneller de data naar het AP gestreamed kan worden, hoe eerder het block verstuurd kan worden.

Daarnaast is 133k absurd groot voor een plaatje voor de tags! Voor een 4.2" display (400x300) is de maximale grootte exact 30.000bytes, 400*300/8*2 voor een BWR plaatje. (+ BMP header) Als het monochrome B/W is, is het nog maar 15k. Voor een 2.9/1.54" tag is dat zo'n 9 respectievelijk 6k, geloof ik.

De oneindige retry moet nodig gefixed :)

nlimper, 2023-01-21:
Makes sense. Ik ga me even verder in het bmp-formaat verdiepen. Het is inderdaad een beetje overkill om 24 bits per pixel te verzenden. ;-)

closed 2023-01-21 by nlimper

jjwbruijn, 2023-01-21:
https://raw.githubusercontent.com/VstudioLAB/ZBS_Flasher/main/custom-firmware/Wireless/Sources/Dmtry_s_original/einkTags_0001/dmitrygr-eink/imgTools/bmp2grays.c

Deze utility kan een bmp omzetten naar iets waar je de tag rechtstreeks iets mee kan :)

example:
./bmp2greys 1bppR < drake.bmp > drake-out.bmp