OpenEPaperLink/OpenEPaperLink #17: Crash during content creation / drawing?

jjwbruijn, 2023-02-18:
Something breaks during an update, started since the version that introduced RSS feeds

```Connected! IP address: 10.0.15.108
start reading DB from file
92
finished reading file

assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

Backtrace:0x40083745:0x3ffd55600x4008dea9:0x3ffd5580 0x400933b1:0x3ffd55a0 0x4008ee5d:0x3ffd56d0 0x400e1147:0x3ffd5710 0x400d9e4d:0x3ffd5970 0x400da77d:0x3ffd5aa0 0x400daeca:0x3ffd5ae0

  #0  0x40083745:0x3ffd5560 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402
  #1  0x4008dea9:0x3ffd5580 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:128       
  #2  0x400933b1:0x3ffd55a0 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x4008ee5d:0x3ffd56d0 in xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1549 (discriminator 1)
  #4  0x400e1147:0x3ffd5710 in wsLog(String) at src/web.cpp:149
  #5  0x400d9e4d:0x3ffd5970 in drawNew(unsigned char*, bool, tagRecord*&) at src/contentmanager.cpp:75 (discriminator 5)
  #6  0x400da77d:0x3ffd5aa0 in contentRunner() at src/contentmanager.cpp:45
  #7  0x400daeca:0x3ffd5ae0 in timeTask(void*) at src/main.cpp:28```

jjwbruijn, 2023-02-18:
Perhaps related, a WDT timeout during RSS feed creation with a corrupted backtrace:
```
RSS feed
E (345491) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (345491) task_wdt:  - IDLE (CPU 0)
E (345491) task_wdt: Tasks currently running:
E (345491) task_wdt: CPU 0: timed tasks
E (345491) task_wdt: CPU 1: IDLE
E (345491) task_wdt: Aborting.

abort() was called at PC 0x40109a30 on core 0

Backtrace:0x40083745:0x3ffbea2c |<-CORRUPTED

  #0  0x40083745:0x3ffbea2c in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402

ELF file SHA256: 0000000000000000

Rebooting...```

nlimper, 2023-02-19:
fixed in #18 
It happened during startup: Websockets were called before they were initialised. Order of initialisation fixed.

closed 2023-02-19 by nlimper