OpenEPaperLink/OpenEPaperLink #25: Spacing on PCB for DevKitC board

sammachin, 2023-03-31:
Hi,

You were kind enough to give me one of your PCBs at Hackaday in Berlin, I've ordered a new ESP-S3-DevKitC-1 genuine espressif board and the header pins, just getting to assemble things.
I've noticed that the spacing on the PCB is a bit too wide for the ESP32 board, I think I can just make it fit with angling the pins slightly but thought I'd let you know for any revisions.
![IMG_6402](https://user-images.githubusercontent.com/187645/2
[image]
29098209-dc471d80-f3ad-4850-b5b1-57434a4426ef.JPG)

jjwbruijn, 2023-04-03:
Interesting! :) And nice fix, of course.

This is what mine looks like:
[image]

Got it from here:
https://www.ebay.com/itm/255671518333

jjwbruijn, 2023-04-03:
Also, please note that that specific board requires a botch-wire; from the top of resistor R5 to GPIO0 on the ESP32. Was unrouted, it's fixed in the current revision in the PCB-folder on the repo

closed 2023-04-04 by sammachin

jjwbruijn, 2023-04-04:
Added this info to the PCB Readme file, thanks for the headsup! Also please note that the 2x5 header in your photo seems to be soldered on in reverse.

Sometimes it's possible to wiggle the plastic shroud up from the pins, and fix it like that. Depending on how hot you soldered the pins, there's a slight risk of ripping the pads off.

Some careful dremel'ing can also turn this header into a omnidirectional header ;)

sammachin, 2023-04-04:
Yes I realised I'd got the header the wrong way around with the key after I
did it, fortunately this was the first one I did so I just reversed the
connector on the other PCB too!