OpenEPaperLink/OpenEPaperLink #44: Feature request: Enable roaming between multible APs

jonasniesner, 2023-05-21:
I don't know if this is even possible but would be useful for events with a lot of moving tags.

nlimper, 2023-06-09:
The tag already switches over to a different AP if it cannot communicate with the associated AP for a long time. But for a real roaming experience, the tag should do a scan of the channels much more often, that will drain de battery quickly.
Of course, the timings for determining if the current AP cannot be reached can be set tighter. At the moment it's at least 8x 10 minutes if I understand the code right. @jjwbruijn probably knows the reasons behind that, and the implications of lowering the timings.

jjwbruijn, 2023-06-25:
This is now configurable in the latest tag FW with config options

closed 2023-06-25 by jjwbruijn