OpenEPaperLink/OpenEPaperLink #45: Feature request: Add easy Tag firmware update and Tag firmware information to the websocket output.

jonasniesner, 2023-05-21:
New firmware could be downloaded to the ESP automatically after a firmware update an shown a a prompt to the user (if the current firmware version would be added to the Tag check in websocket packets, I could also add this feature to my HA integration)

jonasniesner, 2023-05-21:
For this feature request, #43 should be implemented first

nlimper, 2023-06-09:
Personally, I don't like prompts like 'there is a new update ready for you', as imho they are mostly just disturbing, and for people that just use the system without any problems, not necessary (as we don't have security updates, due to the fact there's no security anyway ;-). 
But, if you want to do a version check in HA, you can do a GET request to `/sysinfo`, to get the current running version:
```
{
	"alias": "Access Point office",
	"env": "Simple_AP",
	"buildtime": "1686305344",
	"buildversion": "custom",
	"sha": "0",
	"psramsize": 0,
	"flashsize": 4194304,
	"rollback": true
}
```
`buildversion` will contain the current release version if you run an automatic build.
For performing an update, the process relies on javascript in the browser, so initiating a update remotely will not easily be possible. But at least you can inform the user that there is an update available.

nlimper, 2023-06-28:
Added tag fw version and current channel id to the webinterface (and so, also to the websockets stream).
```{
	"tags": [{
		"mac": "00000218CEC43B18",
		"hash": "a242ffcbbeb6505314e0b2b4058f633d",
		"lastseen": 1687951509,
		"nextupdate": 1687952083,
		"nextcheckin": 1687951569,
		"pending": false,
		"alias": "* button",
		"contentMode": 16,
		"LQI": 78,
		"RSSI": -52,
		"temperature": 24,
		"batteryMv": 2550,
		"hwType": 1,
		"wakeupReason": 2,
		"capabilities": 36,
		"modecfgjson": "{\"location\":\"Eindhoven\",\"ttl\":\"30\",\"#lat\":\"51.44083\",\"#lon\":\"5.47778\",\"#tz\":\"Europe/Amsterdam\"}",
		"isexternal": false,
		"rotate": 0,
		"lut": 0,
		"ch": 25,
		"ver": 19
	}]
}```
.

closed 2023-07-04 by jjwbruijn