OpenEPaperLink/OpenEPaperLink #48: Feature request: PNG upload

jonasniesner, 2023-05-23:
Expanding jpg2buffer to also work with PNG's would be great because jpg is not that great for compressing text and often creates unwanted artifacts

nlimper, 2023-05-24:
I tried, but the library and code is too big for the firmware. Besides that, png's are harder to handle by the esp, regarding memory usage, bigger files, and processor power. Luckily, jpeg's are very well capable of doing the job, even without artifacts. :-)

closed 2023-05-24 by nlimper

seeers, 2023-05-27:
i also have problems with jpg, i can't get a nice output without artifacts. ( Python Image / paint ) 
bmp or png would certainly be better

atc1441, 2023-05-27:
As a quick fix you can save the images as monochrom bmp in paint and then save them again as jpg, the red will be gone but the image gets nice

seeers, 2023-05-27:
thank you, that works! 
Now I just need to generate nice text jpgs from python

nlimper, 2023-05-27:
do you have a simple python program that generates a simple jpg 296x128 which shows the problem?

nlimper, 2023-05-27:
I'm not a python programmer, but I asked ChatGPT nicely. After some extra convincing about my intentions, it came up with this working example:
[jpg_demo.zip](https://github.com/jjwbruijn/OpenEPaperLink/files/11582895/jpg_demo.zip)
I will add the demo to the repo.

seeers, 2023-05-27:
Sure:
```
from PIL import Image, ImageDraw, ImageFont

width, height = 296, 128
img = Image.new("RGB", (width, height), "white")
draw = ImageDraw.Draw(img)
font = ImageFont.truetype("arial.ttf", 35)

text_color = (0, 0, 0)
draw.text((20, height - 99), "Temperatur", font=font, fill=text_color)

temperature_text = "20°C"
text_color = (0, 0, 0)
draw.text((20, height - 60), temperature_text, font=font, fill=text_color)

img.save("temp_testimage.jpg")
```

nlimper, 2023-05-27:
our messages crossed, but: see above.