OpenEPaperLink/OpenEPaperLink #54: blockrequest: couldn't find taginfo 0000000000000000, tags not updating

bartbh, 2023-05-31:
**Describe the bug**
After flashing the lastest version of the esp32 flasher on an esp32 (simple AP version), the tags won't receive the new image. I've tried multiple tags, all with their own, original mac address. In this example I use 0000021C5D4E3B14 as an example.

**To Reproduce**
1. Click tag on webpage
2. Select any of the content options (image, date, weather)
3. Hit save

**Expected behavior**
A new image on the selected tag. As it did in earlier versions.

**Error on webpage**
20:28:10 Updating 0000021C5D4E3B14
20:27:37 Tag not found, this shouldn't happen.
20:27:36 Tag not found, this shouldn't happen.
20:27:36 Tag not found, this shouldn't happen.
20:27:36 Tag not found, this shouldn't happen.
20:26:54 Tag not found, this shouldn't happen.
20:26:53 Tag not found, this shouldn't happen.
20:26:52 Tag not found, this shouldn't happen.
20:26:11 Tag not found, this shouldn't happen.
20:26:11 Tag not found, this shouldn't happen.
20:26:10 Tag not found, this shouldn't happen.
20:25:29 Tag not found, this shouldn't happen.
20:25:28 Tag not found, this shouldn't happen.
20:25:27 Tag not found, this shouldn't happen.
20:25:27 Tag not found, this shouldn't happen.
20:24:40 new image: /current/0000021C5D4E3B14.pending
20:24:40 Updating 0000021C5D4E3B14

**Error in serial console:**
<ADR 0000021C5D4E3B14
blockrequest: couldn't find taginfo 0000000000000000
blockrequest: couldn't find taginfo 0000000000000000
blockrequest: couldn't find taginfo 0000000000000000
<ADR 0000021A0FBF3B1B

**Software versions**
FW version tag: 0.1.5-clut
FW version ap: 0016
version esp32 ap: 30-5 22:55 (CEST) downloaded

jjwbruijn, 2023-05-31:
Thank you for your detailed report!

Good news: There really isn't an awful lot that could go wrong here; I can only see this happening if either the tag sends a block request with a 0-mac, or if the AP-tag somehow zeroes the mac upon receipt.

Bad news, I'm not sure what happens, here. Have you tried force-flashing the AP to the same version again, and maybe reprogram the ESP32 with a flash-erase?

To zero down on the issue, would you mind compiling and running the blockrequest-debug firmware? It'll show a dump of the blockrequest messages as sent by the AP.

https://github.com/jjwbruijn/OpenEPaperLink/tree/blockrequest-debug

It'll barf up messages like these:
```

<RQB file /current/000002A9626E341B.pending block 0, len 4096 checksum 15643
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | 11 CC 9B 71 6B AD A6 A7 09 01 1B 34 6E 62 A9 02 
0x0010 | 00 00
--------------------------------------------------------
<RQB file /current/000002A9626E341B.pending block 1, len 1680 checksum 46205
<XFC 000002A9626E341B
```

bartbh, 2023-06-04:
Hereby the error from the debug  firmware

```
Waiting for valid time from NTP-server
<ADR 00000219B7EA3B1B
+----------------------------+
| AP Information - type 01   |
+----------------------------+
| Channel |             0x0B |
| Power   |               0A |
| MAC     | 00000219B7EA3B1B |
| Version |           0x0016 |
+----------------------------+
AP FW version 0016 - AP_FW_2.9.bin found in FW pack
<ADR 000002187AFC3B1C
finished writing buffer 731ms
>SDA 00000219B7EA3B1B TYPE 0x21
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | EC 50 D2 67 90 6A A7 90 20 00 1B 3B EA B7 19 02
0x0010 | 00 00
--------------------------------------------------------
<RQB file /current/00000219B7EA3B1B.pending block 0, len 4096 checksum 51698
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | ED 50 D2 67 90 6A A7 90 20 01 1B 3B EA B7 19 02
0x0010 | 00 00
--------------------------------------------------------
<RQB file /current/00000219B7EA3B1B.pending block 1, len 4096 checksum 25347
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | EE 50 D2 67 90 6A A7 90 20 02 1B 3B EA B7 19 02
0x0010 | 00 00
--------------------------------------------------------
<RQB file /current/00000219B7EA3B1B.pending block 2, len 1280 checksum 18564
<XFC 00000219B7EA3B1B
<ADR 0000021A0FBF3B1B
ws[/ws][1] connect
ws[/ws][1] disconnect: 0
ws[/ws][2] connect
<ADR 0000021A7FF93B1E
DB saved 123ms
finished writing buffer 273ms
>SDA 0000021C5D4E3B14 TYPE 0x21
<ADR 000002187AFC3B1C
<ADR 0000021B49793B12
<ADR 0000021A0FBF3B1B
<ADR 0000021C5D4E3B14
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | DA 50 D2 67 90 6A A7 90 20 00 00 00 00 00 00 00
0x0010 | 00 00
--------------------------------------------------------
blockrequest: couldn't find taginfo 0000000000000000
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | 6A 50 D2 67 90 6A A7 90 20 00 99 F7 00 00 00 00 
0x0010 | 00 00
--------------------------------------------------------
blockrequest: couldn't find taginfo 000000000000F799
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | FA 50 D2 67 90 6A A7 90 20 00 99 F7 00 90 00 00
0x0010 | 00 00
--------------------------------------------------------
blockrequest: couldn't find taginfo 000000009000F799
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | 8A 50 D2 67 90 6A A7 90 20 00 99 F7 00 20 00 00
0x0010 | 00 00
--------------------------------------------------------
blockrequest: couldn't find taginfo 000000002000F799
<ADR 00000219B7EA3B1B
<ADR 0000021A7FF93B1E
<ADR 000002187AFC3B1C
<ADR 0000021A0FBF3B1B
<ADR 0000021B49793B12
<ADR 0000021C5D4E3B14
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | DA 50 D2 67 90 6A A7 90 20 00 00 00 00 00 00 00 
0x0010 | 00 00
--------------------------------------------------------
blockrequest: couldn't find taginfo 0000000000000000
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | D2 50 D2 67 90 6A A7 90 20 00 FF FF FF FF FF FF 
0x0010 | FF FF
--------------------------------------------------------
blockrequest: couldn't find taginfo FFFFFFFFFFFFFFFF
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | D2 50 D2 67 90 6A A7 90 20 00 FF FF FF FF FF FF 
0x0010 | FF FF
--------------------------------------------------------
blockrequest: couldn't find taginfo FFFFFFFFFFFFFFFF
<ADR 00000219B7EA3B1B
<ADR 000002187AFC3B1C
<ADR 0000021A7FF93B1E
<ADR 0000021A0FBF3B1B
<ADR 0000021C5D4E3B14
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | 62 50 D2 67 90 6A A7 90 20 00 99 F7 00 F8 00 00
0x0010 | 00 00
--------------------------------------------------------
blockrequest: couldn't find taginfo 00000000F800F799
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | D2 50 D2 67 90 6A A7 90 20 00 FF FF FF FF FF FF 
0x0010 | FF FF
--------------------------------------------------------
blockrequest: couldn't find taginfo FFFFFFFFFFFFFFFF
         00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F
--------------------------------------------------------
0x0000 | D2 50 D2 67 90 6A A7 90 20 00 FF FF FF FF FF FF
0x0010 | FF FF
--------------------------------------------------------
blockrequest: couldn't find taginfo FFFFFFFFFFFFFFFF
```

jjwbruijn, 2023-06-09:
Have you tried using the 1.54" AP firmware? This debug log is very interesting :) It's either the AP or a bad tag; since it's at least working for one tag, I suspect one tag that's bad

bartbh, 2023-06-09:
No, not yet. Just found out both of my esp32's we're giving brown out errors at startup. Even with no pins attached and only the esp32 AP firmware.

So I want to swap them with a mini S2 first. To rule out a faulty esp32.

bartbh, 2023-06-11:
Just swapped the doit esp32 for an esp32 s2 and the AP and the tags are working fine now. So I suspect a faulty (or well, 2) esp32, as reverting back to the older version of the esp32 AP didn't resolve the issue.

closed 2023-06-11 by bartbh