OpenEPaperLink/OpenEPaperLink #58: flashing of FW Tag failes "failed to read json header from FW pack"

Uspizig, 2023-06-07:
What a great project.. so far i am still having issue getting it to work.

**Describe the bug**
Can not flash the ZBS243 Tag. Always runs into trouble during flashing.
- triple checked the wiring
- checked with a mulitmeter beeper as well

**To Reproduce**
Steps to reproduce the behavior:
1. Build filesystem image
2. Upload filesystem Image
3. Build and Upload OpenEpaperLink_Mini_AP
4. push reset button on ESP32-s2 mini
5. Watch Terminal output
6. 6. Check littleFS Filesystem in paralell on Mibilephone

`Connected! IP address: 192.168.178.35
reading DB from file
loadDB: Failed to open file
I wasn't able to connect to a ZBS (AP) tag.

This could be the first time this AP is booted and the AP-tag may be unflashed. We'll try to flash it!
If this tag was previously flashed succesfully but this message still shows up, there's probably something wrong with the serial connections.
The build of this firmware expects an AP tag with RXD/TXD on ESP32 pins 6 and 7, does this match with your wiring?
Performing firmware flash in about 30 seconds!

30  29  28  27  26  25  24  23  22  21  20  19  18  17  16  15  14  13  12  11  10  9  8  7  6  5  4  3  2  1  0  MD5=A27D978C48888FA045E8D3F90DC27029
MD5 Matches > Segmented UK
Found an original fw tag, flashing it for use with OpenEPaperLink
Failed to read json header from FW pack
Failed to flash the AP :(
Seems like you're running into some issues with the wiring, or (very small chance) the tag itself
This ESP32-build expects the following pins connected to the ZBS243:
---  ZBS243 based tag              ESP32  ---
       RXD     ----------------     06
       TXD     ----------------     07
       CS/SS   ----------------     11
       MOSI    ----------------     10
       MISO    ----------------     08
       CLK     ----------------     09
       RSET    ----------------     13
Your firmware is configured without soft power control. This means you'll have to manually power-cycle the tag after flashing.
Please verify your wiring and try again!
---- Closed serial port COM10 due to disconnection from the machine ----`
**Expected behavior**
Flashing should run without issues (if D5 Hash can be read out))
Don't get the error message json header from FW pack

**Screenshots**
Filesystem is there will add a screenshot in aminute

atc1441, 2023-06-07:
A check of the size "AP_FW_Pack.bin" would help in the future, 
it will only be 29 bytes big when only linked, so everything under a few Kbytes could seen as wrong so the user can be informed to download the correct one.

I think an forced filesystem OTA update via the Browser should also download the correct one

nlimper, 2023-06-07:
I discussed it with Jelmer already. Now we have OTA in place, we can download the AP firmware in the same way, so the fwpack will not be needed anymore in the near future.

atc1441, 2023-06-07:
maybe its good to have both variants, the current one in the repo as a pack like now and still updates via OTA
That way it should also be ready to run even without any internet connection?!

Uspizig, 2023-06-08:
thanks for the hints. @atc1441 pointed to the correct solution. All I had to do was replace the AP_FW_Pack.bin firmware via the browser.

Now it works like a charm.

@jonasniesner : all I did was doing a "download Zip" via the code button in this repository.
used operation system: Win 10(but shouldn't matter)

closed 2023-06-08 by Uspizig