OpenEPaperLink/OpenEPaperLink #59: Feature request: Support for ICS/ical/(caldav?) calendars

jhbruhn, 2023-06-10:
**Is your feature request related to a problem? Please describe.**
Displaying a Calendar (natively) is currently only possible via Google Calendar and a separate app script.

**Describe the solution you'd like**
Network calendars are often shared via a .ics link, similar to rss feeds. It would be nice to directly integrate those.

**Describe alternatives you've considered**
It probably would be possible to use the Home Assistant Integration and parse the calendar there, but then you could argue the same for RSS feeds.

**Additional context**
Maybe this library can be used? https://reference.arduino.cc/reference/en/libraries/uical/

nlimper, 2023-06-10:
It would be very nice to be able to use ICS url's. I did start using [uICAL](https://github.com/sourcesimian/uICAL) before I build the app scripts method, but ran into issues. Sadly I forgot what the issues were... :-(  I will leave this open, maybe I will try again some day, or somebody else feel like trying to build it. :-)

jhbruhn, 2023-06-13:
Preliminary version published here: https://github.com/jhbruhn/ics-adapter
Docker image is also available there, confirmed working with the nextcloud calendar urls:
```
http://<ip>:3000/get?url=<ics-url>
```

nlimper, 2023-06-13:
Nice. Much better then getting the esp32 to parse a potentially very big calender file. If it's production ready, feel free to add a link to it in the wiki. :-)

jhbruhn, 2023-08-11:
Done: https://github.com/jjwbruijn/OpenEPaperLink/wiki/ICS-calendar

closed 2023-09-27 by jonasniesner