OpenEPaperLink/OpenEPaperLink #61: Feature request: check file before sending ota to tag

bartbh, 2023-06-11:
I was stupid enough to sending a faulty firmware.bin to 2 of my tags. 1 was the AP firmware instead of the TAG firmware, the other was sending the AP_FW_Pack.bin to the tag.

Both resulted in the tag not communicating to the AP and stuck with the message "AP Mode" or "Updating!". This can hopefully be fixed by pulling out my hw-flasher and flashing new, correct firmware.

In order to prevent this from happening to others (or myself again), maybe a simple filecheck could help? Before a .bin file is transferred to a tag.

jjwbruijn, 2023-06-11:
Yeah it's certainly fixable by flashing the tags again, so that's good. We want to add an option to do updates with files retrieved automatically from github, but that'll take some rewriting. That should help in the future :)

On the other hand, I don't like validation all that much. I want to able to brick my tag OTA :)

jonasniesner, 2023-09-27:
The new file names should make it even more unlikely to upload the wrong file. I'm closing this for now but keep it in mind for possible future automatic tag ota.

closed 2023-09-27 by jonasniesner