OpenEPaperLink/OpenEPaperLink #65: Feature request: OTA with own, custom .bin

bartbh, 2023-06-15:
Is it possible to upload a custom firmware (with different pinout) to the esp32 flasher via OTA?  Either by uploading it directly or via url. 

The current update feature works very well, but only when using the pre-defined pinout. Or maybe storing the pin layout in a different location so it doesn't get overwritten?

nlimper, 2023-06-15:
Just for my information, what’s your reason for using a different pinout? We have precompiled environments for esp32, esp32-s2 and esp32-s3. Does your hardware differ from those?

jjwbruijn, 2023-06-15:
The pinout is fixed, or baked in to the code, at compile time. Making it flexible would require a lot of extra clock cycles for every pin change, so that's not a great idea. Building an extra OTA method will costs extra space, which we don't have a lot of.

As @nlimper stated, there already are a bunch of supported environments, I would kindly suggest to change your pin mapping to match one of the existing environments and go from there, or make your own version with custom URL OTA support

closed 2023-06-15 by jjwbruijn

jonasniesner, 2023-06-16:
If you think there is a valid reason to use a different pinout, you can create a PIO config for it and it can be included in the CI/CD pipeline