OpenEPaperLink/OpenEPaperLink #74: No UTF-8 in "Current weather"

DL5SMB, 2023-06-26:
I noticed that in "Current Weather" non-ASCII characters in city names disappear.
Cities like "Köln" and "Tübingen" are being found and weather data is shown, but on the tag they appear as "Kln" and "Tbingen".

I also found that if a city name is entered with the first letter in lowercase like "stuttgart", it is also shown in lowercase on the tag. I assumed that city data would either be matched against an official list or that the first letter was being forced in uppercase.

nlimper, 2023-06-26:
Did you use the latest font files in /fonts? The accented characters were added last month. If you’re still missing a character, I can add it. 

Displayed upper/lower case is by design, it’s just displayed exactly the way you enter it. if you want first upper case, just enter it like that. Remember, it’s running on a 2,5 euro microcontroller, and aimed to be used by electronics hobbyists. I did not have complete foolproofness in mind when developing it. 😃 And we’re very low on flash memory space, so parameters are not 100% checked.

DL5SMB, 2023-06-26:
I git pulled the latest version on friday I think. So I am maybe 1 or 2 commits behind - but as the correct characters are shown in RSS feeds, I think its specific with the code of the weather function.

Yep, I got that. It's no complaint, it was just an observation and I was not sure if that was the supposed behaviour.

nlimper, 2023-06-26:
By the way, proper casing of place names is not a matter of converting everything to lower case, and make the first character upper case. Think of
- ‘s-Hertogenbosch
- Den Haag
- Bad Schandau
- Bad Neustadt a. d. Saale

On which size display are the characters lacking? (It’s not a utf8 problem, but if any, it’s the lack of rendering specific characters in specific fonts)

DL5SMB, 2023-06-26:
I'm with you - that was just my first or better second thought. As I mentioned before, I assumed the entered data was matched with a database to get the coordinates of a location. So I thought there should be a field with the offical name. Not important.

I only tried the 1.54 display yesterday, assuming it was missing everywhere. But I can also test 2.9 tags tonight.

DL5SMB, 2023-06-26:
Okay, so I updated my SimpleAP to the latest version and retried:
UTF-8/ Non-ASCI characters in city names do work on 2.9 inch Tags, but not on 1.54 inch tags.
I can also test segmented displays, but first I need to built a JIG.
[image]

nlimper, 2023-06-27:
Fixed in commit https://github.com/jjwbruijn/OpenEPaperLink/commit/5376f2540e31b89134efadd3f0b1a829a176ebf3
Segmented displays can hardly display characters. At no point the location or any other user input is displayed, so no need to fix.

closed 2023-06-27 by nlimper