OpenEPaperLink/OpenEPaperLink #76: Feature request: If i sent you new Solum ones based on nRF52811 would you be able to do anything with them?

Lordomus, 2023-06-30:
I have a few of the new BWR/BWY ones that are based on nRF52811 (2.2" and 2.6" M3 Newton), and after disassembling one i saw that they changed the SoC to nRF52811, they run on Bluetooth BLE and have integrated NFC Capabilities, but i lack (for now) tools to make any sense of them, I can see few exposed pads, probably you can get JTAG out of them. Would you be interested in those?

jjwbruijn, 2023-07-01:
Thank you for your kind offer!

The nRF52811 is a very capable SoC and it sounds like a pretty cool toy to play around with. As it supports 802.15.4, in theory, it can be made to work with OpenEPaperLink, but it's going to cost a pretty significant amount of time to write a new firmware, as the current (8051-based) firmware is somewhat specific to the underlying hardware. Being a Nordic product, the SoC itself is very well documented, but you'll need to find out how the stock firmware talks to the EPD.

Below are the steps required, these are the two main choices:

Primary route:
1. Take detailed photo's or scans of the PCB, or use a multimeter to trace out the connections from the SoC to the Eink-display.
2. [Reverse engineer the EPD setup routines](https://www.youtube.com/watch?v=gGNzAxeF2o4). You'll need to find out what kind of EPD controller we're working with here.

Secondary route:
1. Glitch [APPROTECT](https://www.youtube.com/watch?v=tMPD0kBG_So) so the SoC will spill its secrets
2. Dump the flash
3. Poke at the dumped firmware with Ghidra until you know how the EPD is initialized, and which pins are used.

After this, you'll need to use the information you found in the steps above to write your own firmware, compatible with OpenEPaperLink, or use BLE instead. This is probably what will cost the most time, especially if (like me) you're not awfully familiar with the soc-family and toolchain.

While I'll happily unburden you from these tags, I cannot promise you when, if at all, I will get around to writing a compatible firmware for these tags. Being relatively new compared to previous generation we currently use with this project also means that they're not yet widely/cheaply available on the second hand market, which as of right now makes them a somewhat less interesting target for me.

Maybe there are other people here with more experience who like a challenge :)

jjwbruijn, 2023-07-17:
@atc1441 wrote a version for the nRF52811!

closed 2023-07-17 by jjwbruijn