OpenEPaperLink/OpenEPaperLink #81: Feature request: Add second checkin after NFC or GPIO wake

jonasniesner, 2023-07-09:
For some applications it could be beneficial to initiate a second checkin for new content after a GPIO/NFC wake.This way externally generated content(for example from Home Assistant) could quickly be displayed after a button press without waiting for the next regular timed window. This feature could be implemented without changing the tag firmware. The delay between second and initial checkin should be configurable to be able to adopt to different external content generation systems

nlimper, 2023-07-09:
I change in tag fw is needed for this. The checkin-delay time that can be communicated to the tag is in minutes, with ‘0’ being no extra delay, so 40 seconds. But I agree, it would be nice to have a shorter delay after gpio/nfc/rf-wake up. 

In addition: imagine a new datatype to load an image without displaying it. That way you can pre-load an image, and quicky display it (using the md5 hash) after a user initiated wake-up.

jonasniesner, 2023-07-09:
sounds like a nice idea, in theory it could be instant by allowing the user to assign an md5 to different "actions" like wakeup by rf/gpio/nfc or maybe even a no ap event or a timer or even on boot. This preloading could be used for so many things...  And maybe also could be invoced over rf as a "display image with md5 of..."

jjwbruijn, 2023-09-27:
Done in 2.2

closed 2023-09-27 by jjwbruijn