OpenEPaperLink/OpenEPaperLink #82: Feature request:  generated "current date" raw images combined into single one

coelner, 2023-07-10:
**Is your feature request related to a problem? Please describe.**
I run out of space. I'm showing the current date on a lot of tags. Each will consume 9.2Kb

**Describe the solution you'd like**
create just one picture for the date and reuse it.

**Describe alternatives you've considered**
use more APs

coelner, 2023-07-10:
38 tags

nlimper, 2023-07-11:
Thanks for submitting.
I don't see a way to do reliable deduplication of the tag images without making things very complicated. But I just added a way to turn off the preview images for the webpage. That way, the .raw-files are not stored on flash anymore after a tag is updated.
Commit https://github.com/jjwbruijn/OpenEPaperLink/commit/b178994bd6015352ab50866ed17f7207b27f2e89 , it will also end up in the next release.

closed 2023-07-11 by nlimper

coelner, 2023-07-11:
Top, that is good enough. Another thing: do you check for free space before rendering images? Maybe it is uncorelated, but it generates a lot of empty files and maybe broken images.

coelner, 2023-07-17:
I have to reopen it. With the change of the date the raw images gets generated and fill up the space. 
There is a need for a scheduler to generate another image if enough space is left. The solution to just generate a single "current date" image and push those to all clients breaks because we can't create symlinks. The "current date" selection is so far the only thing, which is used commonly, the tag mac is used as primary key. as you said, there is no easy way for simplifying it.

https://github.com/jjwbruijn/OpenEPaperLink/blob/ecdadf8c473f84dcb930755cf0f2284089768a55/ESP32_AP-Flasher/src/makeimage.cpp#L213
before the file handler something like this:
```
if (Storage.freeSpace() < bufferSize) {
    wsErr("low on flash. No image generated!");
    return;
}
```
However this failing had to get back to the top and needs to be scheduled again. Maybe I dig up some time to implement a pull request.