OpenEPaperLink/OpenEPaperLink #83: Feature request: ESP32-C6/H2 based AP

g5pw, 2023-07-13:
I'd like to try and contribute an implementation of the AP protocol based only on ESP32-C6 or H2, since both models have a 802.15.4 capable radio.

Is anyone working in something similar? Is the protocol documented somewhere?

Thanks!

jonasniesner, 2023-07-14:
Hi, I wanted to do exactly the same thing (C6 based AP), but did not get around to it. So I would be glad if you implemented it. I started documenting the protocol here https://github.com/jjwbruijn/OpenEPaperLink/wiki/RF-protocol . The esp an Tag code also is a great starting point. If you need more information, you can DM me on Discord or ask your questions here.

g5pw, 2023-07-16:
Thanks! Let's leave this issue open as reference for discussion (I don't use Discord), and I'll open a pull request as soon as I have something to show. I don't have much free time, sadly, but I'll try to wrap it up before end of year; I have all the hardware I need.

jjwbruijn, 2023-07-17:
I'll order a C6 in anticipation ;)

jonasniesner, 2023-08-02:
@atc1441 ported the AP firmware to the C6

closed 2023-08-02 by jonasniesner