OpenEPaperLink/OpenEPaperLink #86: Feature request: Support for Chroma devices? (Easier to find online)

victorhooi, 2023-07-17:
I'm wondering - would you consider adding support for the Chroma range of devices?

These (Chroma) devices seem to be a lot easier to find online (e.g. eBay) than Solum devices - I'm basically not able to find any Solum devices for easy purchase anywhere online.

Dmitry has a section in his writeup on the Chroma74 model (i.e. 7.4")

https://dmitry.gr/?r=05.Projects&proj=29.%20eInk%20Price%20Tags#_TOC_955eca57c5dfe3d5e7de284fb60142e1

There's a bunch of sizes:

https://www.delfi.com/solutions/chroma-red

jjwbruijn, 2023-07-17:
Somewhat intermittingly working on this! There is a small issue; the Chroma's use Sub-ghz radio's, which means it's not going to be as-is compatible with the normal OpenEPaperLink firmware or accesspoints. Also, the Chroma's use SoC's with even more challenging specs than the ZBS243 (half the xram), so don't expect this to happen overnight :) If you want support fast, run these puppies on dmitry's FW!

nlimper, 2023-07-21:
The Solum devices are available for purchase online on Tindie :-)

victorhooi, 2023-07-21:
@nlimper Ah - any chance you'd add shipping to Australia?

nlimper, 2023-07-21:
I got the request. :-) I will calculate.

victorhooi, 2023-07-27:
Ah got it - I think it's letting me checkout now - but the APs are out of stock now? =(

nlimper, 2023-07-27:
They are in stock again :-D
I will close this issue, to keep a bit of tinyness in the issues list. The possible support of Chroma devices will not depend on this very issue. ;-)

closed 2023-07-27 by nlimper