OpenEPaperLink/OpenEPaperLink #91: Feature request: Add another Option JSON Template with Google Script = HTTPS?

Uspizig, 2023-07-23:
Tried to use the the JSON template together with Google script.
This is very convenient for getting various API Data condensed in the Cloud instead of running a Server at home.

Unfortuanetly the JSON Template can right now only deal with HTTP adress.(Not with HTTPS which is needed for Google Script)
If you try to run the below mentioned Google script you run into an error code:
17:27:07 Ok, saved
17:26:45 http https://script.google.com/macros/s/xxxxxxxxxxxxx/exec 302
17:26:43 Updating 0000026D6CFC3B19

Example Script
`
function getEventsForNextDay() {
  
  var events = [{"text": [5,5,"Bahnschrift 20","fonts/bahnschrift20",1]}];
  return JSON.stringify(events);
}

function doGet() {
  var content = getEventsForNextDay();
  var output = ContentService.createTextOutput(content);
  output.setMimeType(ContentService.MimeType.JSON);
  return output;
}
`

Would it be an option to use the same contact mechanism for JSON Template as it is used for Google Calendar?
Feature Request: Add another Option JSON with Google Script?

nlimper, 2023-07-23:
Thanks for the bug report. 
The issue doesn’t seem to be https, but a 302 redirect status. I will look into it, using your test code.

nlimper, 2023-07-24:
Forgot to add to 'follow redirects' parameter, so that's why it ends with status 302. It has nothing to do with https.
I will commit the change in a later commit, but if you already want to use it:
Add this line:
`http.setFollowRedirects(HTTPC_STRICT_FOLLOW_REDIRECTS);`
to this position: https://github.com/jjwbruijn/OpenEPaperLink/blob/master/ESP32_AP-Flasher/src/contentmanager.cpp#L952

closed 2023-07-24 by nlimper

Uspizig, 2023-07-28:
still running into trouble:

22:11:32 InvalidInput
22:11:28 Updating 0000023908653B1A
22:11:27 Ok, saved

nlimper, 2023-07-28:
that's a different error.  :-)
InvalidInput indicates that the json could not be parsed. Open the configured url in your browser (e.g. https://script.google.com/macros/s/xxxxxxxxxxxxx/exec ), en check the syntax. On https://jsonlint.com there is a validator.

Uspizig, 2023-07-28:
Thanks for high speed response:
already checked the JSON: This is just the above mentioned example script and nothing else:
https://script.google.com/macros/s/AKfycbyrrFfMVeoou5IEPGIktS2BCO-TYBLNUKnAyoWZSECR47zVXvI-Z0ipud3tzmSipGFe/exec

and validated with https://jsonformatter.curiousconcept.com/ as well as your link

nlimper, 2023-07-28:
Thanks for the testlink. I will try to reproduce the error later this evening.

nlimper, 2023-07-29:
Fixed it. It will be in a commit later today.
If you don't want to wait for that: add
`http.useHTTP10(true);` around the same position as the last fix.

Uspizig, 2023-08-06:
Thanks for the Great Work:
To complete the example: Demo Code for Google script:

`
function getEventsForNextDay() { 
  var events = [

  {"text": [5,5,"Wettervorhersage","fonts/bahnschrift20",1]},
    {"box": [10,30,20,20,1]},
    {"box": [35,30,20,20,2]},
    {"triangle": [60,30,60,50,80,40,1]},
    {"text": [5,80,"Plain text glasstown_nbp_tf","glasstown_nbp_tf",1]},
    {"text": [5,95,"Plain text 7x14_tf","7x14_tf",2]},
    {"text": [5,110,"Plain text t0_14b_tf","t0_14b_tf",1]},
    {"text": [135,5,"Bahn30","fonts/bahnschrift30",2]},
    {"text": [215,5,"70","fonts/bahnschrift70",1]},
    {"text": [150,80,"50","fonts/calibrib50",2,0]},
    {"text": [205,60,"80","fonts/calibrib80",2]},
    {"text": [90,35,"calibrib30","fonts/calibrib30",1]},
    {"line": [10,120,290,120,1]},
    {"line": [10,115,290,115,2]}
 ];

  return JSON.stringify(events);
}

function doGet() {
  var content = getEventsForNextDay();
  var output = ContentService.createTextOutput(content);
  output.setMimeType(ContentService.MimeType.JSON);
  return output;
}
`