OpenEPaperLink/OpenEPaperLink #92: Feature request: Control tags simultaneously with the same input

Draexl, 2023-07-24:
In my usecase, all tags should display the same thing. At the moment I can only control one tag with the JSON template upload.
Can this be implemented or is there already a way?

Draexl, 2023-07-24:
I just saw that I can store a JSON URL in the TAG.
I have now stored a JSON in File Explorer. What should the URL be now (in the tag settings)?
I can't get it.
used http://192.168.1.100/test.json

Draexl, 2023-07-24:
OK, I found it. You have to put it in the www folder.
But what doesn't work is the interval function in the tag settings.
I always have to click on save in the tag settings

slimline33, 2023-07-24:
You can send a refresh after Uploading a New Image or json

Draexl, 2023-07-24:
My structure looks like this:
I have the JSON file in the ESP/AP directory (Is adjusted via input mask there)
In the TAG settings I refer to the URL of the JSON file.
You can set one minute in the interval, but it is 10 minutes in the background.
That's why I can't send a refresh. I would have to do that for each DAY individually

nlimper, 2023-07-24:
I have three thoughts on this:
- ?cmd=refresh should not be misused. In normal operation, it makes no sense using it. If you want to use it to re-initiate a pull request, it's better to use http POST to push the image instead.
- trying to refresh the displays very fast will not scale well. ESP32 will get too busy drawing useless images, and the tags will not last long with the battery. Polling every 10 seconds is waaaaay to fast. If you want to poll because the image 'might' have changed, it's better to choose a push method. See https://github.com/jjwbruijn/OpenEPaperLink/wiki/Image-upload
- At the moment, all content of the different tags is independant. The image has to be rendered for every tag individually. Maybe it could be a good idea to be able to set the content of a tags as a copy of another tag, so multiple tags will automatically change if you change the content of the 'source' tag. I will look into it to add that functionality.

Draexl, 2023-07-24:
Point 2. The picture does not change every 10 seconds. More like every 15 minutes and even just for a few hours. After that, they won't be needed again for a few weeks.
It is currently recognized that nothing should be changed:
LOG: "new image is the same as current or already pending image. not updating tag."

Point 3. That would of course be the very best idea if the same thing is to be displayed on all tags.

nlimper, 2023-07-24:
By the time the log says "new image is the same as current or already pending image. not updating tag", the image is already rendered from the json (which takes a few seconds). 

If you can pull the image from your webserver (instead of misusing the esp32 as a proxy webserver ;-) ), you could make use of the 'If-Modified-Since'-header. The esp32 sends this header, filled with the last time the image was fetched. Your webserver could then send a http 304 if your image hasn't changed since that last time. This way, you can still do fast polling, but without unnecessary image rendering.

Or, like mentioned, use http POST to push changed.

I will start working on the 'mirror tag contents from another tag', I estimate to finish that in 1-2 weeks.

Draexl, 2023-07-24:
Great, thanks for the info. I will then wait for your implementation with the Master / Slave

Draexl, 2023-07-27:
A request. Is it possible with the Master Slave implementation that the transmission "optional" works simultaneously? Even if this function requires a lot of battery capacity, no problem. The tags are only used temporarily (no long-term use)

nlimper, 2023-07-31:
The current protocol doesn't allow for some kind of broadcasting the content. So it has to be send individually.

Draexl, 2023-08-01:
OK, understood. I'm definitely looking forward to the master/slave implementation :-)

nlimper, 2023-08-03:
done! You can check it out:
https://github.com/jjwbruijn/OpenEPaperLink/commit/df783c6f8f7f86bd6310e291f5d3af4f5ca6a62d
At a content card, you can choose 'display a copy' and set the mac address of the tag you want to mirror.
A mirrored tag doesn't take any file system space (if the tag is local to the AP). But the data still has to be send individually to the tags (there is no broadcast option in the protocol), so they still update one by one.
Let me know the results. :-)

closed 2023-08-03 by nlimper

Draexl, 2023-08-03:
Thanks alot. How can I install it? Unfortunately, my case is glued and I can't get to the RST button on the esp. Unfortunately, only OTA is then possible.

nlimper, 2023-08-03:
If you plan to ever do some development yourself, you can install vscode and platformio, and compile the source. Using the usb-c port, you can program the esp32-s2 in your access point without pressing any of the buttons (that's only needed the first time when you start with an ompty esp32-s2). 
But if you're not planning to ever make changes to the source yourself, installing vscode and platformio is a bit overkill. In that case, I would wait until the next release, which should be in a few days. The changes will be there, and you can use OTA to update.

Draexl, 2023-08-06:
Yesterday I switched to 0.8. But could not find a display a copy on the content card.

Today, I switched back to 0.7. (Completely reinstalled). But apparently no OTA update is currently possible anymore. Is this known?

nlimper, 2023-08-06:
what is the exact message you're getting?

nlimper, 2023-08-06:
About 'But could not find a display a copy on the content card' -> look in the /www folder. If you see two files 'content_cards.json', and 'content_cards.json.gz', delete 'content_cards.json' and leave the .gz one (you can do that with all duplicate files). I could not automatically delete it, because of the risk of bricking if the update was stopped halfway (the delete process runs before the download process). Next release these files will be automatically deleted.

Draexl, 2023-08-06:
> what is the exact message you're getting?

It works... did you do something? It didn't work all day...

Draexl, 2023-08-06:
> About 'But could not find a display a copy on the content card' -> look in the /www folder. If you see two files 'content_cards.json', and 'content_cards.json.gz', delete 'content_cards.json' and [...]

I check it

Draexl, 2023-08-06:
i do it, but i can't i see nothing

[image]
[image]

nlimper, 2023-08-06:
Hm :-( For some reason the updated file didn't end up in the release. Use this one attached.
[content_cards.json.gz](https://github.com/jjwbruijn/OpenEPaperLink/files/12269773/content_cards.json.gz)

Draexl, 2023-08-06:
Works, Thanks :-)