OpenEPaperLink/OpenEPaperLink #93: Change SSID and Wifi Passwort

Draexl, 2023-07-24:
Can I change the SSID and WiFi password?
I have it all running in my test environment. Unfortunately, I can no longer reach the reset button because I glued the case together.

jonasniesner, 2023-07-24:
You will either have for the next release this weekend or go to a location with known wifi. If the ap can not connect to it's known wifi, it will start up in AP mode.

jonasniesner, 2023-07-24:
If you are using the current code from GitHub, there is a button in the AP config to change the wifi SSID and password

nlimper, 2023-07-27:
Good one. :-) 
The next release (end of this week) will have a new WifiManager, which stays active while the AP is running. From then on, you can change the wifi settings on the /setup page at any time. Updating to the next release will be possible over the air, so luckily you don't have to dig out your reset button :-)

closed 2023-07-27 by nlimper

Draexl, 2023-07-28:
@nlimper Perfect :-) Then I can leave the saw in the workshop.
Kudos to the whole team for the quick implementation.
I'm having a lot of fun with the project.