OpenEPaperLink/OpenEPaperLink #95: Feature request: Faster check whether new input is available

Draexl, 2023-07-28:
Topic was already touched on briefly in #92.
According to my observations, the TAG checks every 60 seconds for a new input.
Would it be possible to shorten this review? Via an external menu item in the settings. Of course I am aware that this is not good for the battery life, but in my case it would be perfectly OK.
Possibly with a warning you should set a value less than 60 seconds

nlimper, 2023-07-31:
The tag checks in every 40 seconds, but a bit slower during congestion or bad reception.
There's currently no way in the protocol to change the check in time by software.
In theory, you could change it and compile your own tag firmware. The value is located here: https://github.com/jjwbruijn/OpenEPaperLink/blob/master/zbs243_Tag_FW/powermgt.h#L19

closed 2023-07-31 by nlimper

morfeus2, 2023-10-25:
I add a detail, since I was looking to obtain the same result, faster checkin: it's not enough to reduce INTERVAL_BASE, you must also tweak the infopage, but since I was not able, I ended up commenting the following 2 lines:

    // check if we should sleep longer due to an override in the config
    //if (avg < tagSettings.minimumCheckInTime) return tagSettings.minimumCheckInTime;
   
in the file powermgt.c

Draexl, 2023-10-25:
Can you describe in more detail what this does?
You only comment on two lines.

morfeus2, 2023-10-25:
it checks the value of _minimumCheckInTime_ stored in infopage and I suppose that it finds 40 s, so even if you set _INTERVAL_BASE_ to something like 10 s, you still get a checkin frequency of once every 40 s. If you comment that two lines, it does not care about what is stored in infopage and follows what you put in _INTERVAL_BASE_