OpenEPaperLink/OpenEPaperLink #97: X Y Coordination "Center" specification in template

Draexl, 2023-08-04:
Json template: 
It would be great if it were possible to specify the X Y coordinates as the center. So that the alignment remains the same for different long texts.

nlimper, 2023-08-04:
you can already add the horizontal alignment in the last parameter of the text:
`{ "text": [x, y, "text", "fontname", color, alignment] }`
0: left, 1: center, 2: right
Vertical alignment is always 'top', as the baseline of a font is not always the same, and the perceptual center of gravity of a line of text is not always in the middle.

Draexl, 2023-08-04:
Thanks for the Info. But with die Small Letter e.g. t0_14b_tf is center not possible

nlimper, 2023-08-04:
ah, I see. I will add it soon!

nlimper, 2023-08-05:
Thanks for noticing. Fixed in the next commit.

closed 2023-08-05 by nlimper