OpenEPaperLink/OpenEPaperLink #98: Remote tag update causes AP tag to reboot

rglenn, 2023-08-05:
**Describe the bug**
Whenever a tag starts to update (Block Request received message is shown in log), the AP tag reboots, and the remote tag fails to update.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on a tag
2. Draw a new picture, and hit Save
3. Wait for tag to check in

**Expected behavior**
Remote tag's display should update

**Additional context**
AP tag display blanks and returns to "run". Channel displayed on AP tag resets from selected value (27 in my case) to 11.
AP reports version 0017, tags came pre-flashed with 0.1.9-VER. ESP32-S2 was programmed with [install website](https://install.openepaperlink.de/) at about 1:00 AM GMT on August 5, 2023. Tried with two tags (2.9" and 1.54")

rglenn, 2023-08-05:
I have now seen one or two cases where the channel displayed on the AP tag does not reset, so that appears to be intermittent

rglenn, 2023-08-05:
I think I know what's happened: I didn't see the assembly video, so I never soldered the VCC jumper in place. So it was probably back-powering the tag through GPIOs, which worked until the AP tag tried to transmit a large packet.

Unfortunately, my attempts to re-flash the tag through AP_force_flash.bin have corrupted the info block and the flash memory, so I need to figure out how to re-flash the tag properly. But it's not a bug, I'd say.

closed 2023-08-05 by rglenn

nlimper, 2023-08-05:
Glad you found the cause, and thanks for the insight! 
I will pre-tin the VCC jumper on all flex pcb's from now on. I have to admit, although I've designed that jumper on the pcb myself, I also forgot it the first time. 
Regarding flashing: I'm don't think the info block is needed on the AP tag (that's just to hold the mac on the displays running the tag fw). Using the AP_force_flash.bin, copied from AP_FW_Segmented_UK.bin, should do the trick.