OpenEPaperLink/OpenEPaperLink #99: Feature request: Image file upload from AP web interface

linusg, 2023-08-07:
Unless I missed some hidden option, it seems like it's currently not possible to upload an image file from the AP web interface; only to draw one onto the provided canvas. Preparing an image in a proper editor and exporting as a correctly sized JPG allows for much more control of course, so it'd be nice to allow selecting a file from disk and uploading it to a tag via the AP.

As a workaround I did the following:

```console
$ base64 -i image.jpg
```

and then in a JS console on the webpage after opening the canvas-based image editor:

```js
const image = new Image();
image.src = "data:image/jpg;base64,<paste output from above>";

const canvas = document.querySelector("canvas");
const ctx = canvas.getContext("2d");
ctx.drawImage(image, 0, 0);
```

...and then upload as if it was hand-drawn.

linusg, 2023-08-07:
I forgot to mention that I did see the instructions in https://github.com/jjwbruijn/OpenEPaperLink/wiki/Image-upload, but they are not nearly as convenient as a file picker in the browser :)

nlimper, 2023-08-07:
Thanks for the suggestion. Yes, very useful addition. I will put it on my list.

nlimper, 2023-08-12:
Done! You can now just drag/drop a file to a content card on the overview screen. Any image and any resolution will work, it is converted automatically by the browser.
https://github.com/jjwbruijn/OpenEPaperLink/commit/3472b5d4b6b756a2f64e310103f4d643d80dfcda
available on the next release.

closed 2023-08-12 by nlimper

jhahvhi, 2026-10-12:
https://github.com/jhahvhi/easy-openepaperlink-image-editor

Try this utility 😉