OpenEPaperLink/OpenEPaperLink #100: Encrypted Direct Wifi Connection

Draexl, 2023-08-08:
It would be great if you could secure the Wifi direct connection with WPA.
Encryption is currently only possible in connection with a router.

Draexl, 2023-08-18:
Is this possible? Or a more detailed description is required.

nlimper, 2023-08-18:
Yes, it's definately possible. It can do WPA2, probable it's as simple as setting a password when enabling the wifi AP (and changing the config page to be able to set it...). But to be honest, it's very low on my priority list. If anyone would like to add this: feel free!

Draexl, 2023-08-18:
Good to know it's possible.
The topic may be too specific for that. But for me it would be the perfect solution, since I wouldn't need a router then. I transfer the JSON template using a tablet.
Configuration interface would not be that important and now the stupid question... currently I can't activate this anywhere in the code that is in File Explorer?

Draexl, 2023-08-21:
I think if you tackle that then you immediately make the option that the SoftAP settings password are secured at the same time.
But... I did some research. Apparently there is no way to encrypt the SoftAP with WPA. At least I didn't find anything under "Wifimanager" in the GITHUB documentation.

nlimper, 2023-08-21:
"Wifimanager" is just the name of a library. We don't use that library anymore, I've written my own wifi manager to replace it, because the original 'wifimanager' library was huge, in terms of flash usage.
esp32 is very well capable of doing WPA2, among others. See the documentation of [wifi_auth_mode_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html?#_CPPv416wifi_auth_mode_t) and [wifi_cipher_type_t](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html?#_CPPv418wifi_cipher_type_t) at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html?#_CPPv416wifi_ap_config_t

Draexl, 2023-08-21:
Ah, thanks for the information.
If I were to take the short route now and hardcode the whole thing, would you only have to make the changes here?

void WifiManager::startManagementServer() {
    if (!_APstarted) {
        Serial.println("Starting configuration AP, ssid OpenEPaperLink");
        WiFi.mode(WIFI_AP);
        WiFi.softAP("OpenEPaperLink", "", 1, false);
        WiFi.softAPsetHostname("OpenEPaperLink");
        IPAddress IP = WiFi.softAPIP();
        Serial.printf("IP Address: %s\n", IP.toString().c_str());
        _APstarted = true;
        _nextReconnectCheck = millis() + _retryIntervalCheck;
        wifiStatus = AP;
    }

nlimper, 2023-08-21:
yes, probably you can just change one line:
`WiFi.softAP("OpenEPaperLink", "", 1, false);`
to
`WiFi.softAP("OpenEPaperLink", "MyPassword", 1, false);`

Draexl, 2023-08-22:
Works great. thanks

closed 2023-09-27 by jonasniesner