OpenEPaperLink/OpenEPaperLink #129: Feature request: Hostname should be AP (alias name)

kroon040, 2023-09-09:
**Is your feature request related to a problem? Please describe.**
I notice in my router that the name for that AP is esp32s2-4B40B8. I should be nice like other projects (tasmota, esphome, etc) that you can set the hostname in the interface.

**Describe the solution you'd like**
The name you set in the webinterface that it also should be the hostname on the network

**Describe alternatives you've considered**
setting hostname  with dhcp

**Additional context**

nlimper, 2023-09-09:
thanks for the suggestion. I added the DHCP hostname. (a dhcp hostname is only set during a dhcp request, so you need to reconnect your AP if you change the alias to view the change)
https://github.com/jjwbruijn/OpenEPaperLink/commit/2e64ed2f16815d05c8882836ffcc9d259066abb9
Will be added to next release.

closed 2023-09-09 by nlimper

kroon040, 2023-09-09:
I prefer not to set in trough dhcp, but just take the name in the webinterface and than set through something like this

https://randomnerdtutorials.com/esp32-set-custom-hostname-arduino/

nlimper, 2023-09-09:
Did you take a look at https://github.com/jjwbruijn/OpenEPaperLink/commit/2e64ed2f16815d05c8882836ffcc9d259066abb9 ? It does exactly that, alongside some filtering of illegal characters that can't be used. It sets the alias you configures in the webpage, as the dhcp hostname.

kroon040, 2023-09-09:
Didn't look, checked and looks fine. Thnx for adding