OpenEPaperLink/OpenEPaperLink #130: Feature request: MQTT messages on display

kroon040, 2023-09-10:
**Is your feature request related to a problem? Please describe.**
MQTT client connection to broker, and subscribe to specific topics

**Describe the solution you'd like**
It should me nice that you can post from tasmota mqtt values in json right to the AP and you can select the values you want to display on the screen

**Describe alternatives you've considered**
Now you have an Homeassistant or something with it

**Additional context**

jonasniesner, 2023-09-27:
This might be hard to realise for all APs because of flash limitations. So if somebody would implement it, it would be a feature not available on all APs. So it might be better to use an additional System like Home assistant for now.

jonasniesner, 2023-12-30:
I'm closing it as not planned for now. Feel free to reopen it or to implement this feature yourself if it is still required.

closed 2023-12-30 by jonasniesner

victorhooi, 2024-09-03:
Hmm, I was looking at some mqtt stuff recently as well, and then came across this.

I'm curious - I thought MQTT was quite lightweight, and designed for low-powered IOT-type applications.

Or do you still need quite a bit of memory for MQTT usage? (I haven't done much ESP/Arduino dev work)

From a cursory search, I saw there's quite a few Arduino libraries for MQTT - e.g.:

https://github.com/256dpi/arduino-mqtt

And I think the ESP IDF includes a MQTT client as well?

Or is the idea also more steering people towards:

https://github.com/OpenEPaperLink/Home_Assistant_Integration