OpenEPaperLink/OpenEPaperLink #133: Feature request: invert colors

orichienal, 2023-09-20:
**Is your feature request related to a problem? Please describe.**
no

**Describe the solution you'd like**
it would be nice if you could invert the displayed, e.g. at the weather data

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

nlimper, 2023-09-21:
nice idea. I can work on that. It can take a while before I implement it, but it's on my list. :-)

nlimper, 2023-09-28:
done
https://github.com/jjwbruijn/OpenEPaperLink/commit/db80d23b52cf1899e9c4401b4c2b36419fd9ceb4

closed 2023-09-28 by nlimper