OpenEPaperLink/OpenEPaperLink #138: Feature request: Better documentation

Derek-K, 2023-09-26:
Looking for better documentation on various topics, here are some feedback.

"Flasher"
ESPFlasher seems to be the universal one, but it can only be ESP32-S2. Is it possible to have a modified version use a regular ESP32 module with USB-Serial (and remove Serial0 or SoftwareSerial for debug port)?
ZBS_Flasher only works with ZBS243/SEM9100 8051 based MCU, but it can be ESP8266/ESP32/Aduino Nano
Links to precompiled binaries for the flasher would be a great addition.

"Building an Access Point"
It's great to show the high level "AP Comparisation table", but there are some missing pieces.
Under "Radio" row, does 2.9" tag include all 2.9" tags or just the ZBS243/SEM9100 8051 based MCU?
Wiring diagram for each AP?
Firmware for the ESP32 module (it would be nice to include a link to the binary) for different version of AP (Kudos for the web flash)
Is there a circuit diagram for YellowAP/MiniAP? And why 2 ESP modules is needed? Why can't a single ESP32-C6 being the radio and controller?

Binary FW, under release->assets
For example, AP_FW_2.9bin and Tag_FW_2.9bin, which MCU is for?
Any FW for nRF52811 based tags?
One would assume there is a specific FW for different MCU (and perhaps screen size) of each model of tag, perhaps including model # as part of the filename would be a better option than just the size? (e.g. 2.9", there are various models on Wiki)
What's in AP_FW_Pack.bin?
What is OpenEPaperLink_AP_and_Flasher.bin and what's extra in the "full" version?

Thanks in advance for those who can help clear up these questions.

jonasniesner, 2023-09-26:
The firmware binary situation is already getting worked on an I have an almost ready solution for it. I will push it later today. The ESP32 firmware bundled with every release comes in two flavours. The _full version is used for the web flasher and bundles thing like the bootloader into a single file. The other one is just the firmware for OTA. If you use the web installer and ota, you should not need to do anything manually with them.

jonasniesner, 2023-09-26:
For the AP situation: 2 ESPs are needed because the esp32-C6 has problems with concurrent Wifi and IEEE 802.15.4 and since historically the AP and ESP where always seperated. You could use an NRF based display as an AP but it is not recommended, because this config is not regularly updated. 

The flasher is a new project and was just started a bit ago so still work in progress. Feel free to write a PIO config for other target boards. An Arduino Nano will not work though. If you want to install the flasher on an esp32-S2 you can use install.openepaperlink.de

nlimper, 2023-09-26:
The 2.2" nRF52811-based tags are perfectly usable as tags. Binaries are at https://github.com/jjwbruijn/OpenEPaperLink/tree/master/ARM_Tag_FW/Newton_M3_nRF52811
(for now... we will locate them to a better place in the future).
To use them as an AP: no. It's far more convenient to use a esp32-c6 devkit, or build/get a yellow AP or Mini AP v3, which all are far more powerful.

nlimper, 2023-09-26:
Added pinouts (https://github.com/jjwbruijn/OpenEPaperLink/wiki/Access-point-pinouts) and made some clarifications on the access point page (https://github.com/jjwbruijn/OpenEPaperLink/wiki/Building-an-Access-Point).
Renaming/moving of the tag firmware files is almost ready and will take place tomorrow (https://github.com/jjwbruijn/OpenEPaperLink/pull/139)

The wiki is editable for everybody, so feel free to make additions or clarifications as needed.

closed 2023-09-26 by nlimper