OpenEPaperLink/OpenEPaperLink #144: TTF font rendering not perfect for some fonts

B0rax, 2023-10-13:
**Describe the bug**
If fonts are rendered very big, smooth curves may be rendered incorrectly

**To Reproduce**
Steps to reproduce the behavior:
1. use the font "Universe LT Std 49 light ultra condensed" (I attached it here [Univers.ttf.zip](https://github.com/jjwbruijn/OpenEPaperLink/files/12889867/Univers.ttf.zip))
2. Send Print big characters (like text size 400). This is the json I used: [{"text":[26,5,"12.10.","fonts/Univers.ttf",1,0,400]}]
3. This is the result:
[image]

**Expected behavior**
This is how it looks rendered on macOS:
[image]

As you can see, the corners are rounded incorrectly on the OEPL image.
Here is an overlay of both images:
[image]

**OEPL AP used**
```
env:                ESP32_S3_16_8_YELLOW_AP
build date:         2023-10-08 15:55
esp32 version:      2.01b
filesystem version: 2.01b
psram size:         8341447
flash size:         16777216
--------------------------
```

B0rax, 2023-10-14:
Thanks for the pointers! Don't get me wrong, I think you did an amazing job with the truetype rendering. I just wanted a place to document my findings.

I used the analysing page you linked (font drop) to look how the characters of this particular font are made. For example, here is the top of the "0" character that shows the artefacts:
[image]

This shows me that for this character, we have very long curves.
I looked into the `generateOutline` function. The calculation of the points on the spline is done here: https://github.com/jjwbruijn/OpenEPaperLink/blob/c4beaa51c8171153472ed257c041242610317010/ESP32_AP-Flasher/src/truetype.cpp#L768-L769

The calculations looked fine to me, but I plotted them in Geogebra to look at them. (maybe that is not the most efficient way to do it, but it works).  This is the result and it looks like the spline is correctly calculated:
[image]

So the calculations of the intermediate points are fine. I will look into it more.

nlimper, 2023-10-15:
Thanks for checking. It's nice to investigate the cause together.
Is suspect there might be some amplified rounding issues somewhere, however I cannot pinpoint them yet.

The outline is generated in the `generateOutline` function, but it is not actually drawn. If you activate the commented out parts of the `addLine` function, the outline is actually drawn. If you then comment out call to the `fillGlyph` function in line 953, only the outline is generated, without a fill, which maybe give a bit clearer view. Next step I would do is to Serial.print the actual coordinates passed to the addLine function to see if they would make sense.

nlimper, 2023-10-22:
Found it!
you were very close... It doesn't help to change the starting point during the drawing of the curve... x0 and y0 are needed to draw the line between the intermediate points on the curve, but of course it should not be used in the bezier formula as they change with every line segment.
Changed it to `pointsOfCurve[0].x` and `pointsOfCurve[0].y` -> fixed.

Will be soon in a commit in the repo, somewhere in the coming days. Thanks for diving in to this together.

closed 2023-10-22 by nlimper

B0rax, 2023-10-22:
Nice! Thank you!

nlimper, 2023-10-22:
[image]
[image]

before and after :-)
Fixes in commit https://github.com/jjwbruijn/OpenEPaperLink/commit/a4e19b19ab78218cd3e6251c9de081c3aafeeb5a