OpenEPaperLink/OpenEPaperLink #147: A lot of “white squares” could be printed with unset vars

svenove, 2023-10-15:
**Describe the bug**
When using JSON-templates with variables, the variable name is printed if it has no assigned value. If using large fonts, they don’t contain the letters needed to print the variable name. This causes the "Unicode white boxes" to be printed. 

**To Reproduce**
Steps to reproduce the behavior:
1. Use a JSON-template with '{my_var}' in font 'calibrib80'. 
2. Don’t set the variable to a value. 
3. See the tag print white squares.

**Expected behavior**
Either nothing should be printed instead of the variable, or a char like '?', '-', or similar. 

**Screenshots**
[image]

**Additional context**
I might try fixing this myself and adding it to a PR.

nlimper, 2023-10-15:
Thanks for the report.
I will replace a non-existing variable by an empty string to avoid the uglyness. I cannot assume that a '?' character is present in the font. The fix shows up in the next upcoming commit.
And, if you apply a json template containing a variable, you should make sure that you also fill the variable anyway ;-)

closed 2023-10-15 by nlimper

svenove, 2023-10-15:
Wow, you’re quick!

Yeah, '?' Would require adding it to all the fonts. That’s why I thought about '-', since that was just added in all of them to support negative numbers? If a custom font is used that doesn’t have '-', it would result in a single white box. 

Some variables are set just daily. After a reboot of the AP that would give boxes for quite a while until the var is set. ;)