OpenEPaperLink/OpenEPaperLink #155: Geocoding problem

kvandt, 2023-11-04:
When selecting a location on Current Weather and Buienradar content, I can only type the name of the place. Since I live in Soest (in The Netherlands) I enter "Soest". The geocoding ends up with Soest in Germany. I cannot change this by entering "Soest NL", "Soest Utrecht" or whatever variant. For Buienradar content, I managed to change the Lat/Lon values in the tagDB.json and they remain, so this is a relatively simple manual correction. When doing the same for Current weather, the geocoding is continuously redone (because the Lat Lon values keep being overwritten for Soest Germany).

**To Reproduce**
1. Create a Current weather tag. Select "Soest" for location
2. Check tagDB.json. -> Lat: 51.57558, Lon: 8.10619, which is Soest Germany.
3. Edit tagDB.json . -> Lat: 52.17333, Lon: 5.29167, which is Soest Netherlands
4. Save
5. Wait for some minutes
6. Check tagDB.json again. Lat/Lon values have been overwritten by German Soest values again.

**Expected behavior**
I would expect the geocoding to be done only once (saves a lot of API calls to open-meteo). A place does not move... When manually updated Lat/Lon values, these should be kept.
Ideally the Lat/Lon values can be edited from UI. So you enter a place, Lat/Lon values are retrieved from geocoding and presented in the UI, then accept manual change and save.

scottjl, 2023-11-27:
same issue here. would be easier if the latitude and longitude fields weren't locked. i'm in a city in the US with a rather common name, typing in my city name alone gets the wrong state.

nlimper, 2023-12-30:
fixed, unlocked the lat/lon fields
https://github.com/jjwbruijn/OpenEPaperLink/commit/424cf2faf6772dba38a2e1c2f7706bc658536467

kvandt, 2024-01-01:
Hi Nic, I see you fixed this for the weather forecast and current weather, bit not for Dutch buienradar. Can you do the same magic for this content type?

nlimper, 2024-01-01:
done (on my internal development file, it will show up in some later commit).
If you want to change it for your own install: download, unpack and edit /wwwroot/content_cards.json.gz and change `"type": "ro"` to `"type": "text"` in the Buienradar part, that's all.

kvandt, 2024-01-02:
I modified the code, and can now update the Lat/Lon. But every update the Lat/Lon are overwritten by a geocoding call. This geocoding call should only happen once when entering the location.
So unfortunately the fix is not done yet...

kvandt, 2024-01-05:
Thnx. It looks like the cfgobj parameter is not correctly filled (or updated) when entering getLocation. Even when lat / lon are filled, it apparently does meet the `if (util::isEmptyOrNull(lat) || util::isEmptyOrNull(lon)) {` statement and therefore does a new geocoding. Unfortunaltely I cannot debug this.
Must be something between the "save" of the tag config in the webpage, and entering the first update of the buienradar/weather/forecast.

nlimper, 2024-01-05:
no need to debug it, I know exactly what the problem is. :-)
The #lat and #lon are hidden fields, they are not saved at the front end side. But also, I want to make a nicer solution.
But anyway, for now, just will in a more unique place name of a location nearby.

kvandt, 2024-01-05:
Clear on the hidden fields... But does this also explain why it does not work when I edit the lat and lon values in the tagDB.json file on the ESP? The first update of the tag still triggers a geocoding call.

I know use Soestdijk, which works for me but does not look nice when living in Soest ;-)

nlimper, 2024-01-06:
tagDB.json is just a copy of the actual database that is in memory, which is saved every 5 minutes. If you edit tagDB.json and want to use it, you should pull the power plug after editing, because if you reboot any other way, the databasefile is overwritten with the version from memory.

kvandt, 2024-01-09:
Just tried 2.07b. I love the solution of showing a dropdown to select the "correct" location. For now it seems to work nicely. I will keep an eye on it. Thnx!

nlimper, 2024-01-09:
thanks for letting me know!

closed 2024-01-09 by nlimper