OpenEPaperLink/OpenEPaperLink #159: Access point forgets content of all tags

seeers, 2023-11-09:
I would like to report a possible problem that has now occurred for the second time after a few weeks of operation. 
The access point "forgets" the set content of all tags. Nothing has been changed for a few days, everything was forgotten overnight. 
The last image is still displayed on the tags, in the access point all tags are set to "Static image" as shown in the screenshot. Of the 5 tags, three were configured with a json template, one with weather and one with the current date. 
env:                OpenEPaperLink_Mini_AP
build date:         2023-10-09 14:10 (compiled from github master branch on this date) 
esp32 version:      custom
filesystem version: unknown
psram size:         2090543
flash size:         4194304

Status since this morning: 
[image]

seeers, 2023-11-10:
Another thing that occurs to me is that a tag has disappeared from the Currently active tags. Here the battery is empty, maybe it has something to do with that.

JacSavage, 2023-11-12:
I can confirm this happening to my two tags also.

seeers, 2023-11-13:
did you also run out of battery on a tag and did it then disappear from the list?

marcovannoord, 2023-11-16:
I also have this problem. I have not been able to catch it red-handed when doing this, but i have seen this pop up:
```C++
sleeping for 2 more minutes
>SDA 1234567890ABCDEF NOP
ACK>
ADR><ADR 1234567890ABCDEF
ADR><ADR 0000026B83793B13
ADR><ADR 0000021F7D533B19
ADR><ADR 0000027B20433B12
DB saved 312ms
ADR><ADR 000002BD0BC93B13
Maximum Continuous Heap Space: 10740

abort() was called at PC 0x4018fd7b on core 1

Backtrace: 0x40083889:0x3ffb3810 0x4008e3c9:0x3ffb3830 0x40093ee1:0x3ffb3850 0x4018fd7b:0x3ffb38d0 0x4018fdc2:0x3ffb38f0 0x4018fd23:0x3ffb3910 0x401901e2:0x3ffb3930 0x40190a0d:0x3ffb3950 0x400e2f11:0x3ffb3970 0x400e32b9:0x3ffb39f0 0x400ded71:0x3ffb3a50 0x400e0145:0x3ffb3fe0 0x400e08b1:0x3ffb4150 0x400e2a52:0x3ffb41c0 0x4010eb81:0x3ffb4290

  #0  0x40083889:0x3ffb3810 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4008e3c9:0x3ffb3830 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40093ee1:0x3ffb3850 in abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c:46
  #3  0x4018fd7b:0x3ffb38d0 in __cxxabiv1::__terminate(void (*)()) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
  #4  0x4018fdc2:0x3ffb38f0 in std::terminate() at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
  #5  0x4018fd23:0x3ffb3910 in __cxa_throw at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
  #6  0x401901e2:0x3ffb3930 in operator new(unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_op.cc:54
  #7  0x40190a0d:0x3ffb3950 in operator new[](unsigned int) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/new_opv.cc:32
  #8  0x400e2f11:0x3ffb3970 in spr2color(TFT_eSprite&, imgParam&, unsigned char*, unsigned int, bool) at src/makeimage.cpp:112 (discriminator 1)
  #9  0x400e32b9:0x3ffb39f0 in spr2buffer(TFT_eSprite&, String&, imgParam&) at src/makeimage.cpp:243
  #10 0x400ded71:0x3ffb3a50 in drawBuienradar(String&, ArduinoJson::V6212PB::JsonObject&, tagRecord*&, imgParam&) at src/contentmanager.cpp:1157
  #11 0x400e0145:0x3ffb3fe0 in drawNew(unsigned char const*, bool, tagRecord*&) at src/contentmanager.cpp:402
  #12 0x400e08b1:0x3ffb4150 in contentRunner() at src/contentmanager.cpp:47
  #13 0x400e2a52:0x3ffb41c0 in loop() at src/main.cpp:164
  #14 0x4010eb81:0x3ffb4290 in loopTask(void*) at C:/Users/Marco/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: 75a57596f7de560a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
>
[WiFi-event 0] 
[WiFi-event 2] 
```
It might or might not be related, since this didn't cause the tags to be gone. However, if the system panics while it is writing the database, it might corrupt it.

marcovannoord, 2023-11-17:
I think i managed to catch it in the act:
It tries to write to spiffs, but does not have enough memory, and probably fails.
Would it be sufficient to check the mallocs before trying to write?
``` c++
pingACK> ok, 2ms
ADR><ADR 1234567890ABCDEF
ADR><ADR 000002BD0BC93B13
pingACK> ok, 3ms
time succesfully synced
DB saved 175ms
ADR><ADR 000002BD0BC93B13
ADR><ADR 1234567890ABCDEF
ADR><ADR 0000021F7D533B19
ADR><ADR 000002BD0BC93B13
ADR><ADR 0000027B20433B12
ADR><ADR 0000021F7D533B19
ADR><ADR 1234567890ABCDEF
E (1285829) esp_littlefs: dir struct could not be malloced
E (1285843) esp_littlefs: dir struct could not be malloced
E (1285871) esp_littlefs: dir struct could not be malloced
E (1285884) esp_littlefs: dir struct could not be malloced
E (1286359) esp_littlefs: dir struct could not be malloced
E (1286372) esp_littlefs: dir struct could not be malloced
E (1286400) esp_littlefs: dir struct could not be malloced
E (1286413) esp_littlefs: dir struct could not be malloced

abort() was called at PC 0x4018fd7b on core 1

Backtrace: 0x40083889:0x3ffb3ec0 0x4008e3c9:0x3ffb3ee0 0x40093ee1:0x3ffb3f00 0x4018fd7b:0x3ffb3f80 0x4018fdc2:0x3ffb3fa0 0x40190119:0x3ffb3fc0 0x401901d4:0x3ffb3fe0 0x40193f23:0x3ffb4000 0x40193fee:0x3ffb4030 0x4010135a:0x3ffb40b0 0x401ab6af:0x3ffb40f0 0x400e9726:0x3ffb4110 0x400e9755:0x3ffb4130 0x400e088a:0x3ffb4150 0x400e2a52:0x3ffb41c0 0x4010eb81:0x3ffb4290

ELF file SHA256: 75a57596f7de560a

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
>
[WiFi-event 0] 

```

marcovannoord, 2023-11-22:
Update: 
I have increased the intervals at which the different tasks run, and i have not seen this issue any more for the past 4 days, while normally it would happen every x-hours.

``` c++
// main.cpp:26
util::Timer intervalContentRunner(seconds(2));
util::Timer intervalSysinfo(seconds(9));
util::Timer intervalVars(seconds(60));
util::Timer intervalSaveDB(minutes(60));
util::Timer intervalCheckDate(minutes(35));
```
Apperantly, there can be a race-condition that causes this bug.

kopierschnitte, 2024-01-24:
Same problem here ... happens almost every day.

nlimper, 2024-01-24:
If you use VScode/platformIO, try to install the latest commit, which will probably make the network connection a lot more stable. (this will also be added to the next release, if you just want to update the AP without hassle).
For now, I'm closing this. If it re-appears after the next update, feel free to open it again, but please, no 'me too!' comments. If a bug is there, it will be there for everybody ;-) Of course, if you have new info, it is always welcome to add.

closed 2024-01-24 by nlimper