OpenEPaperLink/OpenEPaperLink #160: Feature request: Add option to perform partial update instead of full refresh

DrBlokmeister, 2023-11-10:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Currently the tags perform a complete refresh where the pixels switch from black to white repeatedly. I assume this costs quite some power, even though it might not be always needed. I found on a Waveshare black/white display that a full refresh every 60 updates works completely fine and almost no ghosting is visible. Please allow for the option to change the frequency of complete refreshes, or to trigger a full refresh separately.

**Describe alternatives you've considered**
None.

**Additional context**
Add any other context or screenshots about the feature request here.

jjwbruijn, 2023-11-10:
Cool, most of the displays here aren't Waveshare though :) Even worse: In many cases there is -no- information available whatsoever. We use shortened LUT's on most M2 displays, but so far I haven't been able to find enough information about other displays to enable stuff like partial refreshes. What we know about many of the displays is based only on what the stock firmware writes to them. Based on the commands that are being sent we can ascertain that the controllers are *similar* to somewhat documented controllers, but not the same. Stock doesn't do partial refreshes, so until we get better documentation, I don't know how to do it.

For now I'm going to close this feature request with a resounding 'nope'. Not because I don't want to, but I don't have enough information to implement this on all but maybe one or two tag types. Sorry. Feel free to reopen if you have enough information to make this happen!

closed 2023-11-10 by jjwbruijn

DrBlokmeister, 2023-11-12:
Thanks for the explanation. Unfortunate but I understand. Can indeed be closed.