OpenEPaperLink/OpenEPaperLink #165: raw generation fails

cmsj, 2023-11-21:
**Describe the bug**
I'm setting up an AP from Nic on Tindie which I got along with some of his tags.

The AP came with 2.01b already installed, but I'm having trouble getting it to send any images to tags.

Looking at the logs, and using the ESP Editor filesystem browser, it looks like the `.raw` image buffer files are empty. Here is a log from me configuring a tag in the AP to show the current date:

```
14:33:36 File has size 0. //00000283A43D3B10.raw
14:33:36 {"errMsg":"File has size 0. //00000283A43D3B10.raw"}
14:33:36 Updating 00000283A43D3B10
14:33:36 {"logMsg":"Updating 00000283A43D3B10"}
14:33:35 Ok, saved
14:33:35 {"tags":[{"mac":"00000283A43D3B10","hash":"00000000000000000000000000000000","lastseen":1700577189,"nextupdate":0,"nextcheckin":1700577249,"pending":false,"alias":"Tag1","contentMode":1,"LQI":46,"RSSI":-36,"temperature":20,"batteryMv":2600,"hwType":1,"wakeupReason":0,"capabilities":4,"modecfgjson":"{}","isexternal":false,"apip":"0.0.0.0","rotate":0,"lut":0,"invert":0,"ch":11,"ver":19}]}
```
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

nlimper, 2023-11-21:
- first, try some simple content, like 'count hours'. Does that work?
- if you try to upload your own jpg, make sure it meets the specifications: exactly the right resolution (296x128), baseline (non-progressive), full color (not grayscale).

cmsj, 2023-11-21:
Same thing if I switch it to Count Hours:

```
20:34:11 File has size 0. //00000283A43D3B10.raw
20:34:11 {"errMsg":"File has size 0. //00000283A43D3B10.raw"}
20:34:11 Updating 00000283A43D3B10
20:34:11 {"logMsg":"Updating 00000283A43D3B10"}
20:34:10 Ok, saved
20:34:10 {"tags":[{"mac":"00000283A43D3B10","hash":"00000000000000000000000000000000","lastseen":1700590660,"nextupdate":0,"nextcheckin":1700600599,"pending":false,"alias":"Tag1","contentMode":3,"LQI":68,"RSSI":-33,"temperature":21,"batteryMv":2600,"hwType":1,"wakeupReason":0,"capabilities":4,"modecfgjson":"{\"counter\":\"1\",\"thresholdred\":\"\"}","isexternal":false,"apip":"0.0.0.0","rotate":0,"lut":0,"invert":0,"ch":11,"ver":19}]}
```

cmsj, 2023-11-21:
I connected the AP to my computer and used install.openepaperlink.de to get the console, and when I hit Save on the tag I see:

```
DB saved 90ms
Failed to allocate buffer
Maximum Continuous Heap Space: 139252
```

cmsj, 2023-11-21:
I reflashed it just in case, using the ESP32_S3_16_8_YELLOW_AP firmware, still seeing the same thing. Here is a boot log - maybe there is a fault on my hardware?

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x2a (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
E (235) opi psram: PSRAM ID read error: 0x00000000, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (235) spiram: SPI RAM enabled but initialization failed. Bailing out.
>
[WiFi-event 0] 
[WiFi-event 2] 
Stored ssid: chrul
[WiFi-event 3] 
[WiFi-event 0] 
Connecting to WiFi...[WiFi-event 2] 

[WiFi-event 4] Connected to access point
[WiFi-event 7] Obtained IP address: 10.0.88.58
Connected!
reading DB from file
loadDB took 252ms
```

nlimper, 2023-11-21:
Yes, looks like a hardware issue. I see you opened an issue on the AP-v3 github too. I'm answering there, closing this one.

closed 2023-11-21 by nlimper