OpenEPaperLink/OpenEPaperLink #167: Make NTP-server configurable

ThinkPadNL, 2023-11-23:
It seems that the NTP-server is [hardcoded](https://github.com/jjwbruijn/OpenEPaperLink/blob/220b4ae3e84e55ea78442e39e506e04907b33929/ESP32_AP-Flasher/src/system.cpp#L18). In my firewall i have blocked all internet traffic for my IOT vlan. I see outgoing NTP-requests to internet. I have a NAT-rule to bend them around to my internal NTP-server, but i would prefer to have it configured correctly on the client.

Would it be possible to add this in the wifi settings?

coelner, 2023-12-06:
maybe this would be an option? : https://github.com/espressif/arduino-esp32/issues/2483
```
sntp_servermode_dhcp(1); //try to get the ntp server from dhcp
sntp_setservername(1, "europe.pool.ntp.org"); //fallback server
sntp_init();
```

nlimper, 2024-01-09:
it's a pretty specific situation. I'm not going to work on this, but if somebody wants to add this and test it, I will merge it.

ThinkPadNL, 2024-02-26:
I have sold all my ESL hardware (not enough time to tinker with ih) so i think it is fair to also close this request as i won't be using this project anymore.

VincentKueszter, 2026-03-13:
I have the same issue as the original author and I think that taking NTP servers from DHCP (or making them configurable at least) is a basic feature for any well-behaved network device.
I also noticed that if I lose the internet connection (or block it), the device hammers my local unbound instance with unreasonably frequent requests for the three hardcoded DNS servers  - this could be avoided as well (i.e. by setting a local NTP server).

[image]

(the numbers are per minute)

I would be happy if this ticket could be reopened. I also would be more than willing to test a fix on my device - I tried implementing a fix myself, but that didn't work out.

nlimper, 2026-10-14:
The latest release takes the NTP server from DHCP (option 42) when your network hands one out, so you can point it to your own NTP server in your DHCP server (commit 1964a9e7). Without that option it uses a single server, time.cloudflare.com, instead of the three hardcoded ones. With a single server, the AP waits longer between retries when there is no internet, so it no longer floods the DNS server. The NTP server in use is shown on the new dashboard. Feel free to reopen if this doesn't work for your setup.

closed 2026-10-14 by nlimper