OpenEPaperLink/OpenEPaperLink #170: NanoC6 AP Web installer ?

Ivanverdonck, 2023-11-29:
Which AP should I choose (using Web installer at https://install.openepaperlink.de/) for downloading the firmware to the NanoC6 AP board?

Thanks,
Kind regards

jonasniesner, 2023-12-12:
The C6 can only be used in conjunction with another Esp32, Esp32-s2 or Esp32s3. This esp should be wired up in a way that allows it to update the c6. After you install the firmware to the main esp, it will update the C6 by downloading the newest firmware directly. Because of this, the install page does not offer C6 firmware. If you can think about a usecase for it to offer the firmware anyways, just reply here and I may add it.

closed 2023-12-12 by jonasniesner

Ivanverdonck, 2023-12-15:
Hi Jonas, that the C6 will be flashed by the other esp was clear for me. But not which firmware u need to flag on the esp32-s2. I am going to build this hardware “NanoC6 by ATC1441” [it](https://github.com/jjwbruijn/OpenEPaperLink/tree/master/Hardware/NanoC6%20by%20ATC1441) it uses an esp32-s2. The “Access point pinouts” section [in](https://github.com/jjwbruijn/OpenEPaperLink/wiki/Access-point-pinouts) the wiki mentions 2 firmwares “mini” and “nano” for the esp32-s2 with different pinouts.

Maybe I need to ask this directly to @atc1441: Which firmware did u flash on the esp32-s2 (WEMOS S2-Min)?

kind regards, Ivan.

atc1441, 2023-12-15:
@Ivanverdonck please flash the NanoAP Firmware on the S2-Mini

kvandt, 2024-01-18:
Hi @atc1441 I have build the NanoC6 and flashed the NanoAP firmware on the S2-mini. I do get the webpage, but I think the C6 has not been flashed. I get "failed" on the homepage after initialisation. When I press the "Update ESP32-C6" button I only get the "Flashing ESP32-C6..." message, but nothing happens. How can I get the C6 flashed?

kvandt, 2024-01-18:
Hi @atc1441 thnx for quick reply. I have flashed the Nano_C6 version now through platformIO. AP boots, but still gives "failed". When going to update page it mentions the version is newer than the latest release date (which is true, because I just created it)
`env:                OpenEPaperLink_Nano_C6`
`build date:         2024-01-18 20:24`
`esp32 version:      custom`
`filesystem version: 2.07b`
`psram size:         2093071`
`flash size:         4194304`

I tried pushing the flash C6 button, but again without succes. What am I doing wrong? Do I need to change settings in the platformio ini?

atc1441, 2024-01-18:
It might be possible that the internal flashing does not work for the S2-Mino -> C6.

So please try this one: https://github.com/jjwbruijn/OpenEPaperLink/tree/master/ARM_Tag_FW/ESP32_S3_to_C6_Flasher_Test

Otherwise you need to flash the C6 externally with an USB to UART converter

The S2-Mini C6 Combo is kinda Legacy as you maybe already noticed

kvandt, 2024-01-18:
Do I need to update filesystem as well?

kvandt, 2024-01-18:
I will try.Thanks for your help. What would be the recommended AP now if I do not get the Nano C6 working?

atc1441, 2024-01-18:
The next better one would be the S3 C6 NanoAP https://github.com/jjwbruijn/OpenEPaperLink/tree/master/Hardware/S3_C6_NanoAP%20by%20ATC1441

kvandt, 2024-01-19:
@atc1441 It worked after flashing the [ESP32_S3_to_C6_Flasher_Test](https://github.com/jjwbruijn/OpenEPaperLink/tree/master/ARM_Tag_FW/ESP32_S3_to_C6_Flasher_Test) repo followed by flashing the original [repo](https://github.com/jjwbruijn/OpenEPaperLink/blob/master/ESP32_AP-Flasher/platformio.ini#L518) again. Will it now be possible to use the upcoming updates normally, or do I have to keep using the platformio route?

atc1441, 2024-01-19:
Very nice to hear, 

You need to go the manual route as no uodates will be compiled for the old NanoC6 anymore, sorry avout that. 
I will add a note in the repo of NanoC6