OpenEPaperLink/OpenEPaperLink #172: Feature request:

Fettkeewl, 2023-12-06:
**Is your feature request related to a problem? Please describe.**
Remembering device display resolution 

**Describe the solution you'd like**
Show tag display resolution on AP "Active tags" webpage / Show in device info for HA 

**Describe alternatives you've considered**
Note it down myself

**Additional context**
[image]

jonasniesner, 2023-12-12:
Can you please elaborate on what you want to achieve with this feature request? Do you want the resolution for all tag types?

nlimper, 2023-12-12:
In the webinterface, it’s currently impossible to quickly see the resolution of a connected tag. I already planned to add it somewhere.

nlimper, 2023-12-30:
done https://github.com/jjwbruijn/OpenEPaperLink/commit/424cf2faf6772dba38a2e1c2f7706bc658536467

closed 2023-12-30 by nlimper