OpenEPaperLink/OpenEPaperLink #175: Orphaned Tag??

mattfox27, 2023-12-11:
Is this an orphaned tag?  All my tags are setup not sure what this is?  I cannot remove it

[image]

nlimper, 2023-12-11:
It's some random noise that was picked up and interpreted as a unknown tag. Until know, you cannot delete it from the web interface, that's a bug I want to solve so I keep this one open.
In the meantime, if you want to remove it: go to the file editor in /edit, and change the file /current/tagDB.json to remove the tag. After uploading the changes, power cycle the AP (by pulling the plug), as using the 'reboot' button rewrites the tagDB from memory.

mattfox27, 2023-12-11:
> It's some random noise that was picked up and interpreted as a unknown tag. Until know, you cannot delete it from the web interface, that's a bug I want to solve so I keep this one open. In the [...]

will do...thanks

nlimper, 2023-12-29:
I fixed the web interface, it's now possible again to delete unknown tags by right clicking. (in latest commit; will be included in next release)

closed 2023-12-29 by nlimper

mattfox27, 2023-12-30:
Awesomes...thank you