OpenEPaperLink/OpenEPaperLink #177: Feature request: Option to not adopt newly seen tags

sknebel, 2023-12-13:
It would be nice to be able to tell the AP to only talk to displays it knows already (bonus: or it has been told explicitly to request). 

This has come up a bit in our hackspace with multiple people in parallel working on their projects and tags being grabbed by random APs instead of the one belonging to the same person as the tag. It would also be useful when bringing setups to events (e.g. I bet there's more than one group bringing APs + Tags to CCC-Congress now :tada: )

A good way would be a checkbox somewhere that tells the AP to only talk to tags that it already has in its database, and ignore requests from unknown tags.

A bonus would be a command/API to add a specific tag MAC to the database without it being present, so once the tag appears and the AP sees it it will be accepted anyways. I could see this being more complicated since I'm not sure the way tags are managed is prepared to deal with tags where only the MAC is known but no other data/connection details yet.

nlimper, 2023-12-13:
Nice and useful idea.

jonasniesner, 2023-12-13:
Another useful way this could be implemented is like the teach in function of ZigBee. Press a button in the web interface and for the next 60 seconds, all now tags will be added to the DB. After that, only tags in the DB are talked to.

nlimper, 2024-03-02:
Not so fast ;-)
While with the 'Lock tag inventory' prevents a tag from showing up in your AP, this option currently doesn't prevent the connected C6 from answering to checkins. Therefore, you cannot run two seperate installs on the same channel yet. Once the C6 firmware get a bit more advanced, I planned to solve this as well. So, for that reason, I like to keep this issue open. It will take a while before this will be added though.

nlimper, 2026-10-21:
@sknebel I wrote “it will take a while”, and it did: almost two years. But it’s there now, in AP firmware 3.00 with radio firmware 0024 for the C6/H2. With “Lock tag inventory” on, the radio itself no longer answers tags that aren’t in the AP’s database. 
(This only works with a C6 or H2 radio. With a TLSR radio, the lock works like before: it keeps unknown tags out of the AP, but the radio still answers them)

Closing this one. Thanks for the idea, and for your patience!

closed 2026-10-21 by nlimper