OpenEPaperLink/OpenEPaperLink #179: Feature request: Add access control to web interface

jum0n, 2023-12-14:
**Is your feature request related to a problem? Please describe.**
Security is everyone's problem?  ;)

**Describe the solution you'd like**
Would really like to a password for the web interface to make it more secure.

Calimerorulez, 2023-12-21:
Good idea; I like to add a request for user access control. One of my kids would like to updated his own tag, but I do not want to give him full access to the web interface...

milo526, 2024-01-11:
I believe these 2 feature requests (by @jum0n and @Calimerorulez) are quite different.

Adding a plain password to the UI is fairly trivial.
Adding accounts and being able to assign them to tags is quite a bit more difficult.

Only limiting the web-interface will not provide any real security since the AP's API can still be interacted with in different methods (but might keep someone's kids out of the UI)

I believe it's dangerous to introduce security features that are more show than actual security since they might give people a false sense of security.

I'd love to help think about what is a good tradeoff given the hardware limitations we are dealing with and the fact that this should probably still be considered a "hobby" project.

nlimper, 2024-01-11:
OpenEpaperLink is for hobby, and not designed to be safe. Take the 'open' in OpenEpaperLink very literally. ;-)
- The procol between AP and tags is not encrypted or secured.
- The UDP multicast messages between the AP's are not encrypted or secured.
- The management website is not encrypted.
- De database can be download by anyone.
- All files in the AP file system can be read *and* written by anyone in your network.
So, like @milo526 corretly states. it makes no sense in securing the management website any more than a basic http password. 
Adding a basic password (http basic authentication) can be done easily (https://github.com/me-no-dev/ESPAsyncWebServer?tab=readme-ov-file#serving-static-files-with-authentication). Feel free to add it!

Calimerorulez, 2024-01-12:
Agreed :-)

closed 2024-02-04 by jjwbruijn