OpenEPaperLink/OpenEPaperLink #181: NanoAP Flashed Only Pulse Blue No Wifi

peterchs, 2023-12-18:
I'm trying to set up a nano ap.

I've flashed it via the web form.

Afterwards on boot the S2 mini led pulses blue then goes off.  I see no wifi point available to configure it.  I do not see anything over the serial connection while it boots.  I also tried another S2 which has no tag connected, it does the same.

Any ideas?

S2 minis I bought, think they are correct spec
https://www.ebay.co.uk/itm/195402587878

peterchs, 2023-12-18:
Could have been cheated on the s2 mini - looks like it doesnt have any PSRAM, think the AP firmware needs 2MB.  Presumably 0MB PSRAM would cause the behaviour I describe?

atc1441, 2023-12-18:
Yes no ram will cause problems.
You can change the platformio.ini to compile for the S2 without PSRAM

peterchs, 2023-12-18:
Thanks - closing as hardware issue.

closed 2023-12-18 by peterchs