OpenEPaperLink/OpenEPaperLink #182: Feature request:

nigels0, 2023-12-18:
**Is your feature request related to a problem? Please describe.**
No - its an enhancement of function

**Describe the solution you'd like**
Really, its an enhancement to the draw function - to allow reverse image, so you can have, say white text in a red background - ideally, have an option when drawing text to say what the background is.

nigels0, 2023-12-19:
Nevermind, managed to do it with the following code:

`left, top, right, bottom = draw.textbbox((x,y) , args.line1, font=font)
draw.rectangle((left-2, top-5, right, bottom+5), fill = 'red')`

nlimper, 2023-12-19:
I don’t know which language you are writing the code, but it’s not something within OpenEpaperLink. 😊

closed 2023-12-19 by nlimper

nigels0, 2023-12-19:
Right - it is in python using the pillow library. Would be great to have a background option antway.

nlimper, 2023-12-19:
The text is not antialiased (because the epaper only can display white/black/red without shades). So, if you use the json template, you can just draw a red box, and use white text.