OpenEPaperLink/OpenEPaperLink #183: Corrupted Display for FW Version v0.24 with SSD1619 Based Tags

Jebsta, 2023-12-18:
**Describe the bug**
After updating the Firmware on my ST-GR2900 and ST-GR1600 Tags the content on them sometimes corrupted itself.
I only have those types of tags, so don't know if it only happens on this type of tag.
Some of the tags even "crashed" and never connected with the AP. Only after reset and then only until the next crash.

**To Reproduce**
Steps to reproduce the behavior:
1. Update the Tags to v0.24
2. Wait for the screen to show a new image
3. See the error (maybe just after some refreshes)

**Expected behavior**
It should just show the image/content

**Screenshots**
[image]
[image]
[image]

jonasniesner, 2023-12-31:
@Jebsta There is a test version on the discord that might fix this, Do you want to try it?

jonasniesner, 2023-12-31:
Should be fixed by ab8cb39

closed 2023-12-31 by jonasniesner