OpenEPaperLink/OpenEPaperLink #187: Tags are waiting for data but never update

jelloj1, 2023-12-29:
**Describe the bug**
Tags are connected to mini-AP - I can reboot them from webpage, but they never update with any content - in the logs it keeps saying "blocked request" and the AP will contstantly reboot

**To Reproduce**
Steps to reproduce the behavior:
Try to update a tag
AP reboots constantly
Tag eventually times out

**Expected behavior**
Expect tag to update per the instructional video

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Purchased both tags and mini-AP kit off Tinde - tags seem to have version 1.9 - learned the "battery" method to get them to boot up and they work fine, find the AP, show up in the webbrowser, but say waiting for data.  everythings is soldered correctly on AP (including the jumper) and it seems to update just fine.  Tried different zigbee channels since I have Phillips Hue on channel 20

nlimper, 2023-12-30:
Not enough information. What kind of content are you sending? If you're uploading an image, make sure you use the exact right resolution, jpeg, baseline (no progressive). First try to use a simple content type (like 'count hours').
If it's not working, provide the output of the serial terminal.

jelloj1, 2023-12-30:
I tried sending all the different content, but to get the logs, I just tried to send the current date.   - Here is a screen shot of the web logs and the attached file is the serial log.  
[image]

[esp-web-tools-logs.txt](https://github.com/jjwbruijn/OpenEPaperLink/files/13798911/esp-web-tools-logs.txt)

For reference x131 is the AP and xB1A is the tag I used in this example.

nlimper, 2023-12-30:
This behaviour will show if the segmented tag is not connected to VCC. It is back-fed by the other pins so it works a bit, but the power is not strong enough for actual sending of an image.

You probably forgot to solder the little blob for connecting VCC:
[image]
Or, if the blob is there, the VCC pad is not connected successfully to the segmented tag.

jelloj1, 2023-12-30:
Ok - I did jumper the VCC - but i will poke around and make sure the solder job is good.

jelloj1, 2023-12-30:
That was it - I resoldered everything and it worked - thanks for your help!

closed 2023-12-30 by nlimper