OpenEPaperLink/OpenEPaperLink #189: No red color on v25

jurasjo, 2024-01-01:
**Describe the bug**
After flashing tags with the v25 firmware, there is no red color. What is supposed to be red font is simply black.

**To Reproduce**
I use for example Weather Forecast option, now we have minus temperatures and on v24, temperatures are red and after flashing to v25, the same temp is black.

**Expected behavior**
Red color should be displayed.

psfales, 2024-01-09:
I'm seeing the same thing, I think.    
[image]
[image]
The all black photo is what I get with v25.   The other photo with the red is after switching back to v22 and how it is supposed to look.

Also, if you look in the lower right of the v25 image where it shows the time, you can see faint traces of an earlier value.    I think this might be related to the lack of sharpness that the OP reported. 

However, it's not always consistent.     When I first installed v25, I know that I was able to see the red (for example, when I tested with "Current Date" ) I could see the red.   It was only after running for several days that I realized this particular display was not displaying any red.

jurasjo, 2024-01-09:
Look like M2 v0026 version is out. I have installed it. Red color and sharpeness are back.

jonasniesner, 2024-01-09:
Fixed with v0026

closed 2024-01-09 by jonasniesner