OpenEPaperLink/OpenEPaperLink #194: Provide a way for tag to report firmware version

psfales, 2024-01-09:
**Is your feature request related to a problem? Please describe.**
As far as I know, the only way to determine what firmware version a tag is running is to reboot it and watch for the message that is (very briefly) displayed early in the boot sequence.

**Describe the solution you'd like**
Would it be possible to add a "sensor" (like the temperature or battery level) that would show the current version?

**Describe alternatives you've considered**
Or any other way to "ask" the tag what firmware is installed without rebooting?

jonasniesner, 2024-01-09:
On newer Tag and AP firmware versions, the tag firmware is shown on the web interface.(for ZBS  based M2 and all M3) the only tags not showing the firmware version are M2 7.4" and most likely all new beta releases in the last few days. If your tag can show it and does not show it on the latest AP firmware, assume it is outdated.

closed 2024-01-09 by jonasniesner

psfales, 2024-01-09:
My face is red.  Thanks for pointing that out - I see it now.