OpenEPaperLink/OpenEPaperLink #202: Improve formatting for Google Calendar

psfales, 2024-01-13:
**Is your feature request related to a problem? Please describe.**
If a Google Calendar entry is longer than one line, it wraps around and overwrites the next line.  (See photo for an example)

**Describe the solution you'd like**
Simplest would be just truncate the line at the right edge.  Nicer would be to use two (or more lines)  for the entry if necessary.  (The subsequent lines could be indented so they line of up with the first, rather than start at the left edge)

**Describe alternatives you've considered**
I could try to remember to keep the calendar entries short, but would prefer not to do that

**Additional context**
[image]

peterchs, 2024-02-18:
I truncate the event titles in the google calendar script to work around this wrapping.

Also the date in the top right doesnt adhere to the configured language.json "date_format" its always dd-mm-YYYY for me.

nlimper, 2024-02-18:
Thanks for the report. I will fix it.

peterchs, 2024-02-24:
Latest 2.5beta fixes date formatting, thanks.
Cant comment on the text wrapping events arent long enough.

nlimper, 2024-03-18:
fixed. Long lines get truncated now.
(in upcoming commit/releases)

closed 2024-03-18 by nlimper