OpenEPaperLink/OpenEPaperLink #204: Mini-ap hangs on an almost daily basis

mooseroy, 2024-01-16:
**Describe the bug**
Bought a mini-ap and a bunch of tags from tindie, and for the most part they work perfectly, updating about 10 tags on an hourly basis, and another 15 daily using the homeassistant plugin.   The issue is that it seems like every other day, the mini-ap will simply stop updating tags.  Trying to ping or connect to the mini-ap's web server with the browser fails as well.   I have to physically unplug and plug back in the ap to get it to respond again.  Looking at the log.txt after the hard reboot shows multiple **Reboot. Reason: Panic message** daily, but then a big gap until I do the hard reboot.  

This has happened on all versions sinceI started using the mini-ap and tags.

Was hoping there was more debugging or information I could extract that would help to fix the issue.

**Expected behavior**
System stays online and doesn't hang during the day requiring a hard reboot.

**Additional context**
Version:
env:                ESP32_S3_16_8_YELLOW_AP
build date:         2024-01-09 15:42
esp32 version:      2.07b
filesystem version: 2.07b
psram size:         8379015
flash size:         16777216
--------------------------

Tail end of log:
2024-01-15 11:20:46 http imageUpload 000002cafd854834.jpg
2024-01-15 11:21:02 http imageUpload 000002b842774838.jpg
2024-01-15 11:37:07 Reboot. Reason: Panic
2024-01-15 11:37:08 Connecting to WiFi...
2024-01-15 11:37:10 Connected!
2024-01-15 11:37:00 00004C6E913BDAEC Booting
2024-01-15 12:00:02 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 12:00:19 http imageUpload 0000021a47163b14.jpg
2024-01-15 12:00:31 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 12:00:43 http imageUpload 0000021eee7a3416.jpg
2024-01-15 12:00:54 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 12:01:04 http imageUpload 000002caff024839.jpg
2024-01-15 12:01:16 http imageUpload 000002cafd854834.jpg
2024-01-15 12:01:29 http imageUpload 000002b842774838.jpg
2024-01-15 13:00:00 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 13:00:14 http imageUpload 0000021a47163b14.jpg
2024-01-15 13:00:26 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 13:00:37 http imageUpload 0000021eee7a3416.jpg
2024-01-15 13:00:49 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 13:01:00 http imageUpload 000002caff024839.jpg
2024-01-15 13:01:12 http imageUpload 000002cafd854834.jpg
2024-01-15 13:01:23 http imageUpload 000002b842774838.jpg
2024-01-15 13:17:30 Reboot. Reason: Panic
2024-01-15 13:17:31 Connecting to WiFi...
2024-01-15 13:17:33 Connected!
2024-01-15 13:17:39 00004C6E913BDAEC Booting
2024-01-15 14:00:03 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 14:00:15 http imageUpload 0000021a47163b14.jpg
2024-01-15 14:00:26 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 14:00:37 http imageUpload 0000021eee7a3416.jpg
2024-01-15 14:00:50 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 14:01:02 http imageUpload 000002caff024839.jpg
2024-01-15 14:01:16 http imageUpload 000002cafd854834.jpg
2024-01-15 14:01:28 http imageUpload 000002b842774838.jpg
2024-01-15 15:19:35 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 15:20:35 http imageUpload 0000021a47163b14.jpg
2024-01-15 15:21:36 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 15:22:09 http imageUpload 0000021eee7a3416.jpg
2024-01-15 15:22:24 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 15:22:35 http imageUpload 000002caff024839.jpg
2024-01-15 15:22:46 http imageUpload 000002cafd854834.jpg
2024-01-15 15:22:59 http imageUpload 000002b842774838.jpg
           19:00:01 Reboot. Reason: Power-on
           19:00:02 Connecting to WiFi...
           19:00:03 Connected!
           19:00:09 00004C6E913BDAEC Booting
2024-01-15 19:47:22 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 19:48:11 http imageUpload 0000021a47163b14.jpg
2024-01-15 19:48:22 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 19:48:36 http imageUpload 0000021eee7a3416.jpg
2024-01-15 19:48:55 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 19:49:09 http imageUpload 000002caff024839.jpg
2024-01-15 19:49:22 http imageUpload 000002cafd854834.jpg
2024-01-15 19:49:41 http imageUpload 000002b842774838.jpg
2024-01-15 19:49:56 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 19:50:11 http imageUpload 0000021a47163b14.jpg
2024-01-15 19:50:22 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 19:50:33 http imageUpload 0000021eee7a3416.jpg
2024-01-15 19:50:49 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 19:51:04 http imageUpload 000002caff024839.jpg
2024-01-15 19:51:16 http imageUpload 000002cafd854834.jpg
2024-01-15 19:51:28 http imageUpload 000002b842774838.jpg
2024-01-15 19:51:39 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 19:51:50 http imageUpload 0000021a47163b14.jpg
2024-01-15 19:52:07 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 19:52:16 http imageUpload 0000021eee7a3416.jpg
2024-01-15 19:52:27 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 19:52:41 http imageUpload 000002caff024839.jpg
2024-01-15 19:52:54 http imageUpload 000002cafd854834.jpg
2024-01-15 19:53:05 http imageUpload 000002b842774838.jpg
2024-01-15 19:53:16 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 19:53:27 http imageUpload 0000021a47163b14.jpg
2024-01-15 19:53:41 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 19:53:52 http imageUpload 0000021eee7a3416.jpg
2024-01-15 19:54:03 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 19:54:15 http imageUpload 000002caff024839.jpg
2024-01-15 19:54:27 http imageUpload 000002cafd854834.jpg
2024-01-15 19:54:37 http imageUpload 000002b842774838.jpg
2024-01-15 20:00:03 http imageUpload 0000021f82ad3b18.jpg
2024-01-15 20:00:17 http imageUpload 0000021a47163b14.jpg
2024-01-15 20:00:29 http imageUpload 0000021b6d403b1e.jpg
2024-01-15 20:00:40 http imageUpload 0000021eee7a3416.jpg
2024-01-15 20:00:53 http imageUpload 0000021f2ff63b11.jpg
2024-01-15 20:01:04 http imageUpload 000002caff024839.jpg
2024-01-15 20:01:15 http imageUpload 000002cafd854834.jpg
2024-01-15 20:01:26 http imageUpload 000002b842774838.jpg

jjwbruijn, 2024-01-16:
What happens if you only generate content for tags from within the webinterface, and don't send any jpegs?

mooseroy, 2024-01-16:
I'll set all the tags to show current date in the web interface, and let you know in a day or two if it happens with said setting.

mooseroy, 2024-01-17:
I ran all day yesterday with the 30 of the tags set to show current date, and 4 showing weather, all driven from the web interface, no home assistant involved.  Checked the log file this morning and saw 1 reboot panic:

`2024-01-16 09:27:35 Reboot request by user
2024-01-16 09:27:40 Reboot. Reason: Software
2024-01-16 09:27:41 Connecting to WiFi...
2024-01-16 09:27:42 Connected!
2024-01-16 09:27:47 00004C6E913BDAEC Booting
2024-01-16 09:33:43 Reboot request by user
2024-01-16 09:33:46 Reboot. Reason: Software
2024-01-16 09:33:47 Connecting to WiFi...
2024-01-16 09:33:48 Connected!
2024-01-16 09:33:56 00004C6E913BDAEC Booting
2024-01-16 09:34:01 0000021876E33B1E Network scan
2024-01-16 09:35:23 0000021F86893B1A Network scan
2024-01-16 09:36:00 0000021EF16A3419 Network scan
2024-01-16 09:36:04 0000021D33763419 Network scan
2024-01-16 09:36:44 0000021A47163B14 Network scan
2024-01-16 09:36:50 0000021F489B3B1B Network scan
2024-01-16 09:37:30 0000021FB9303B14 Network scan
2024-01-16 09:37:30 0000021FB9303B14 Network scan
2024-01-16 09:37:43 0000021B74693B1D Network scan
2024-01-16 09:37:54 0000021F2FF63B11 Network scan
2024-01-16 09:37:54 0000021F2FF63B11 Network scan
2024-01-16 09:38:28 0000021EC20E341B Network scan
2024-01-16 09:38:33 0000021F82AD3B18 Network scan
2024-01-16 09:38:48 0000021B01AA341F Network scan
2024-01-16 09:39:05 0000021877023B10 Network scan
2024-01-16 09:39:18 0000021DC9B33415 Network scan
2024-01-16 09:39:24 0000021EF2B0341D Network scan
2024-01-16 09:39:57 00000218A6AB3B1F Network scan
2024-01-16 09:40:02 0000021F774E3B1F Network scan
2024-01-16 09:41:36 0000021EF5373415 Network scan
2024-01-16 09:41:44 0000021EFB76341E Network scan
2024-01-16 09:41:48 0000021F98823411 Network scan
2024-01-16 09:42:18 0000021B6D403B1E Network scan
2024-01-16 09:43:06 0000021FB7EB3B1C Network scan
2024-01-16 09:43:15 000002CAFD854834 Network scan
2024-01-16 09:43:20 0000021EEE7A3416 Network scan
2024-01-16 09:43:41 0000021EF5B4341E Network scan
2024-01-16 09:43:41 0000021EF5B4341E Network scan
2024-01-16 09:43:56 0000021EEBD93B15 Network scan
2024-01-16 09:43:57 000002B842774838 Network scan
2024-01-16 09:44:14 0000021B5D3F341A Network scan
2024-01-16 09:44:48 0000021FBFFE341F Network scan
2024-01-16 09:45:01 000002CAFF024839 Network scan
2024-01-16 09:45:33 0000021F3CE0341B Network scan
2024-01-16 12:35:07 00000222143D341F Network scan
           19:00:01 Reboot. Reason: Power-on
           19:00:01 Connecting to WiFi...
           19:00:03 Connected!
2024-01-16 13:34:19 00004C6E913BDAEC Booting
2024-01-16 16:36:51 Reboot. Reason: Panic
2024-01-16 16:36:52 Connecting to WiFi...
2024-01-16 16:36:53 Connected!
2024-01-16 16:36:47 00004C6E913BDAEC Booting
2024-01-17 04:00:00 Nightly reboot
2024-01-17 04:00:02 Reboot. Reason: Software
2024-01-17 04:00:03 Connecting to WiFi...
2024-01-17 04:00:05 Connected!
2024-01-17 04:00:10 00004C6E913BDAEC Booting
`

mooseroy, 2024-01-18:
The mini-ap still hasn't hung using only web interface driven updates.  I'm going to start pushing 3 of the updates from home assistant to see how long it takes for it to hang, and if there is any correlation between the number of updates from home assistant and the time it takes to hang.

mooseroy, 2024-01-18:
I'm not sure if this helps in any way, but I did have a console logger connected today when it rebooted due to a panic:
```
ADR><ADR 0000021B6D403B1E
I (4239162) RADIO: RX 20
I (4239172) RADIO: TX 41
ADR><ADR 0000021EC20E341B
I (4240882) RADIO: RX 20
I (4240882) RADIO: TX 41
ADR><ADR 000002B842774838
I (4242172) RADIO: RX 20
I (42Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4202c48b  PS      : 0x00060b30  A0      : 0x8202cd6e  A1      : 0x3fcb1540  
A2      : 0x3fc9c4f4  A3      : 0x3fcae5e0  A4      : 0x3d80f386  A5      : 0x3fc9c570  
A6      : 0x02c97bf0  A7      : 0x00ffffff  A8      : 0x00090000  A9      : 0x3fcb1520  
A10     : 0x3fcae5e0  A11     : 0x00000008  A12     : 0x3fcb2c5c  A13     : 0x00000000  
A14     : 0x3fcf1da0  A15     : 0x00000000  SAR     : 0x00000017  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0009000c  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x4202c488:0x3fcb1540 0x4202cd6b:0x3fcb1570 0x4202cd83:0x3fcb1590 0x4202cda1:0x3fcb15b0 0x42023545:0x3fcb15d0 0x42016d9d:0x3fcb1660 0x4201b892:0x3fcb1720

ELF file SHA256: 99bcd87c71d67feb

Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40377ac4
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
>
[WiFi-event 0] 
[WiFi-event 2] 
Stored ssid: ashborn-iot
[WiFi-event 3] 
[WiFi-event 0] 
[WiFi-event 2] 
Connecting to WiFi...
[WiFi-event 4] Connected to access point
[WiFi-event 7] Obtained IP address: 192.168.30.147
Connected!
reading DB from /current/tagDB.json
loadDB took 454ms
pingPanic! Pausing content generation for 30 seconds
ACK> ok, 79ms
I (4248532) MAIN: RDY? In
ACK>I (4248642) MAIN: SCP In
I (4248642) MAIN: Set channel: 15 power: 10
I (4248862) MAIN: NFO? In
ACK>TYP>VER>MAC>ZCH>ZPW>PEN>NOP>ACK>I (4248972) MAIN: HSPD In, switching to 2000000
switched to 2000000 baud
time succesfully synced
ACK>
| AP Info - type C6       |
| Ch   |             0x0F |
| Power|               0A |
| MAC  | 404CCAFFFE421F18 |
| Ver  |           0x0019 |
Free heap: 231976 allocatable: 221172 stack: 12944
DB saved 790ms
sleeping for 5 more minutes
>SDA 0000021FBFFE341F NOP
ACK>I (4250492) MAIN: SDA In
sleeping for 5 more minutes
>SDA 000002CAFD854834 NOP
ACK>I (4250602) MAIN: SDA In
sleeping for 5 more minutes
>SDA 0000021EC20E341B NOP
ACK>sleeping for 5 more minutes
>SDA 0000021F3CE0341B NOP
I (4250712) MAIN: SDA In
ACK>I (4250822) MAIN: SDA In
sleeping for 5 more minutes
>SDA 0000021F774E3B1F NOP
ACK>sleeping for 5 more minutes
>SDA 0000021EEBD93B15 NOP
I (4250932) MAIN: SDA In
ACK>I (4251042) MAIN: SDA In
<ADR 00004C6E913BDAEC
ADR><ADR 0000021F774E3B1F
I (4251662) RADIO: RX 20
I (4251662) RADIO: TX 41
ADR><ADR 00000218A6AB3B1F
```

mooseroy, 2024-01-19:
Turning back on the updating of around 20 tags once per hour caused a panic.  I had a putty log attached all day, and this was at the tail end of the log before the panic.
```
RQB>[0;32mI (36107092) RADIO: RX 42[0m
[0;32mI (36107102) RADIO: TX 27[0m
ACK>[0;32mI (36107192) MAIN: Starting BlkData, 90 ms after request[0m
<RQB file /current/000002CAFF024839.pending block 7, len 1328 checksum 10194
Sending parts: XXXXXXXXXX XXXX...... .......... .......... ..
[0;32mI (36107302) MAIN: Blockdata fully received in 117 ms, 207 ms after the request[0m
[0;32mI (36107312) RADIO: TX 126[0m
[0;32mI (36107312) RADIO: TX 126[0m
[0;32mI (36107322) RADIO: TX 126[0m
[0;32mI (36107322) RADIO: TX 126[0m
[0;32mI (36107332) RADIO: TX 126[0m
[0;32mI (36107332) RADIO: TX 126[0m
[0;32mI (36107342) RADIO: TX 126[0m
[0;32mI (36107342) RADIO: TX 126[0m
[0;32mI (36107352) RADIO: TX 126[0m
[0;32mI (36107352) RADIO: TX 126[0m
[0;32mI (36107362) RADIO: TX 126[0m
[0;32mI (36107362) RADIO: TX 126[0m
[0;32mI (36107372) RADIO: TX 126[0m
[0;32mI (36107372) RADIO: TX 126[0m
[0;32mI (36107382) RADIO: TX 126[0m
[0;32mI (36107382) RADIO: TX 126[0m
[0;32mI (36107392) RADIO: TX 126[0m
[0;32mI (36107392) RADIO: TX 126[0m
[0;32mI (36107402) RADIO: TX 126[0m
[0;32mI (36107402) RADIO: TX 126[0m
[0;32mI (36107412) RADIO: TX 126[0m
[0;32mI (36107412) RADIO: TX 126[0m
[0;32mI (36107422) RADIO: TX 126[0m
[0;32mI (36107422) RADIO: TX 126[0m
[0;32mI (36107432) RADIO: TX 126[0m
[0;32mI (36107432) RADIO: TX 126[0m
[0;32mI (36107442) RADIO: TX 126[0m
[0;32mI (36107442) RADIO: TX 126[0m
[0;32mI (36107452) RADIO: TX 126[0m
[0;32mI (36107452) RADIO: TX 126[0m
[0;32mI (36107452) RADIO: TX 126[0m
[0;32mI (36107462) RADIO: TX 126[0m
[0;32mI (36107462) RADIO: TX 126[0m
[0;32mI (36107472) RADIO: TX 126[0m
[0;32mI (36107472) RADIO: TX 126[0m
[0;32mI (36107482) RADIO: TX 126[0m
[0;32mI (36107482) RADIO: TX 126[0m
[0;32mI (36107492) RADIO: TX 126[0m
[0;32mI (36107492) RADIO: TX 126[0m
[0;32mI (36107502) RADIO: TX 126[0m
[0;32mI (36107502) RADIO: TX 126[0m
[0;32mI (36107512) RADIO: TX 126[0m
XFC><XFC 000002CAFF024839
[0;32mI (36107602) RADIO: RX 25[0m
[0;32mI (36107612) RADIO: TX 24[0m
ADR><ADR 0000021F98823411
[0;32mI (36112232) RADIO: RX 20[0m
[0;32mI (36112232) RADIO: TX 41[0m
jpeg conversion 152x152
finished writing buffer 163ms
ADR><ADR 0000021D33763419
[0;32mI (36113702) RADIO: RX 20[0m
[0;32mI (36113702) RADIO: TX 41[0m
>SDA 0000021EF16A3419 TYPE 0x21
ACK>[0;32mI (36113812) MAIN: SDA In[0m
CORRUPT HEAP: Bad head at 0x3fcb24c0. Expected 0xabba1234 got 0x3fca3324

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40377eb6:0x3fcadf10 0x4037dd39:0x3fcadf30 0x40383d35:0x3fcadf50 0x40383945:0x3fcae080 0x403786ed:0x3fcae0a0 0x40383d65:0x3fcae0c0 0x420c02d9:0x3fcae0e0 0x420c0925:0x3fcae100 0x4202c13e:0x3fcae120 0x4202c150:0x3fcae140 0x4202c4d5:0x3fcae160 0x4202d354:0x3fcae190 0x4202d369:0x3fcae1c0 0x420c3333:0x3fcae1e0 0x420c3490:0x3fcae210 0x420c3a35:0x3fcae230

ELF file SHA256: 99bcd87c71d67feb

Rebooting...
```

mooseroy, 2024-01-20:
Just adding a couple more traces from panics:
```
ACK>[0;32mI (36107192) MAIN: Starting BlkData, 90 ms after request[0m
<RQB file /current/000002CAFF024839.pending block 7, len 1328 checksum 10194
Sending parts: XXXXXXXXXX XXXX...... .......... .......... ..
[0;32mI (36107302) MAIN: Blockdata fully received in 117 ms, 207 ms after the request[0m
[0;32mI (36107312) RADIO: TX 126[0m
[0;32mI (36107312) RADIO: TX 126[0m
[0;32mI (36107322) RADIO: TX 126[0m
[0;32mI (36107322) RADIO: TX 126[0m
[0;32mI (36107332) RADIO: TX 126[0m
[0;32mI (36107332) RADIO: TX 126[0m
[0;32mI (36107342) RADIO: TX 126[0m
[0;32mI (36107342) RADIO: TX 126[0m
[0;32mI (36107352) RADIO: TX 126[0m
[0;32mI (36107352) RADIO: TX 126[0m
[0;32mI (36107362) RADIO: TX 126[0m
[0;32mI (36107362) RADIO: TX 126[0m
[0;32mI (36107372) RADIO: TX 126[0m
[0;32mI (36107372) RADIO: TX 126[0m
[0;32mI (36107382) RADIO: TX 126[0m
[0;32mI (36107382) RADIO: TX 126[0m
[0;32mI (36107392) RADIO: TX 126[0m
[0;32mI (36107392) RADIO: TX 126[0m
[0;32mI (36107402) RADIO: TX 126[0m
[0;32mI (36107402) RADIO: TX 126[0m
[0;32mI (36107412) RADIO: TX 126[0m
[0;32mI (36107412) RADIO: TX 126[0m
[0;32mI (36107422) RADIO: TX 126[0m
[0;32mI (36107422) RADIO: TX 126[0m
[0;32mI (36107432) RADIO: TX 126[0m
[0;32mI (36107432) RADIO: TX 126[0m
[0;32mI (36107442) RADIO: TX 126[0m
[0;32mI (36107442) RADIO: TX 126[0m
[0;32mI (36107452) RADIO: TX 126[0m
[0;32mI (36107452) RADIO: TX 126[0m
[0;32mI (36107452) RADIO: TX 126[0m
[0;32mI (36107462) RADIO: TX 126[0m
[0;32mI (36107462) RADIO: TX 126[0m
[0;32mI (36107472) RADIO: TX 126[0m
[0;32mI (36107472) RADIO: TX 126[0m
[0;32mI (36107482) RADIO: TX 126[0m
[0;32mI (36107482) RADIO: TX 126[0m
[0;32mI (36107492) RADIO: TX 126[0m
[0;32mI (36107492) RADIO: TX 126[0m
[0;32mI (36107502) RADIO: TX 126[0m
[0;32mI (36107502) RADIO: TX 126[0m
[0;32mI (36107512) RADIO: TX 126[0m
XFC><XFC 000002CAFF024839
[0;32mI (36107602) RADIO: RX 25[0m
[0;32mI (36107612) RADIO: TX 24[0m
ADR><ADR 0000021F98823411
[0;32mI (36112232) RADIO: RX 20[0m
[0;32mI (36112232) RADIO: TX 41[0m
jpeg conversion 152x152
finished writing buffer 163ms
ADR><ADR 0000021D33763419
[0;32mI (36113702) RADIO: RX 20[0m
[0;32mI (36113702) RADIO: TX 41[0m
>SDA 0000021EF16A3419 TYPE 0x21
ACK>[0;32mI (36113812) MAIN: SDA In[0m
CORRUPT HEAP: Bad head at 0x3fcb24c0. Expected 0xabba1234 got 0x3fca3324

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40377eb6:0x3fcadf10 0x4037dd39:0x3fcadf30 0x40383d35:0x3fcadf50 0x40383945:0x3fcae080 0x403786ed:0x3fcae0a0 0x40383d65:0x3fcae0c0 0x420c02d9:0x3fcae0e0 0x420c0925:0x3fcae100 0x4202c13e:0x3fcae120 0x4202c150:0x3fcae140 0x4202c4d5:0x3fcae160 0x4202d354:0x3fcae190 0x4202d369:0x3fcae1c0 0x420c3333:0x3fcae1e0 0x420c3490:0x3fcae210 0x420c3a35:0x3fcae230

ACK>[0;32mI (9908132) MAIN: Starting BlkData, 96 ms after request[0m
<RQB file /current/000002B842774838.pending block 7, len 1328 checksum 11358
<ADR 00004C6E913BDAEC
[... 234 lines cut ...]
[0;32mI (289612) RADIO: TX 126[0m
[0;32mI (289612) RADIO: TX 126[0m
[0;32mI (289622) RADIO: TX 126[0m
[0;32mI (289622) RADIO: TX 126[0m
[0;32mI (289632) RADIO: TX 126[0m
[0;32mI (289632) RADIO: TX 126[0m
[0;32mI (289642) RADIO: TX 126[0m
[0;32mI (289642) RADIO: TX 126[0m
[0;32mI (289652) RADIO: TX 126[0m
[0;32mI (289652) RADIO: TX 126[0m
XFC><XFC 0000021D33763419
[0;32mI (289782) RADIO: RX 25[0m
[0;32mI (289782) RADIO: TX 24[0m
[0;32mI (289992) RADIO: RX 20[0m
[0;32mI (289992) RADIO: RX 20[0m
ADR>ADR>[0;32mI (290002) RADIO: TX 41[0m
[0;32mI (290002) RADIO: TX 41[0m
<ADR 00000218A6AB3B1F
CORRUPT HEAP: Bad head at 0x3fcb2870. Expected 0xabba1234 got 0x3fca3324

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40377eb6:0x3fcb0af0 0x4037dd39:0x3fcb0b10 0x40383d35:0x3fcb0b30 0x40383945:0x3fcb0c60 0x403786ed:0x3fcb0c80 0x40383d65:0x3fcb0ca0 0x420c02d9:0x3fcb0cc0 0x420c0889:0x3fcb0ce0 0x4202bd93:0x3fcb0d00 0x420e01bb:0x3fcb0d20 0x4202cbd3:0x3fcb0d40 0x4202cc0f:0x3fcb0d60 0x4202ccf2:0x3fcb0d80 0x4202cd39:0x3fcb0da0 0x4202cd5b:0x3fcb0dc0 0x4202cd83:0x3fcb0de0 0x4202cda1:0x3fcb0e00 0x42023545:0x3fcb0e20 0x42016d9d:0x3fcb0eb0 0x4201b892:0x3fcb0f70

```

nlimper, 2024-01-20:
If you're running platformio/vscode, you can connect the terminal output there. It will automatically decode the stack trace (the 'backtrace' in the crashes, displaying the exact position in the script. I have no clue about the `CORRUPT HEAP` error, I haven't seen it in my devices.
I'm currently improving some parts of the image upload script to make it more robust. But I'm not sure if they will solve your corrupt heap crashes, as I don't know yet what the origin is.

mooseroy, 2024-01-20:
Thanks. Just installed platformio, so will connect up the terminal and see if it provides anything useful.

mooseroy, 2024-01-20:
Here is the backtrace.  Again, thanks for any help.
```
CORRUPT HEAP: Bad tail at 0x3d810eb9. Expected 0xbaad5678 got 0x08ad5678

assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)

Backtrace: 0x40377eb6:0x3fcf5ec0 0x4037dd39:0x3fcf5ee0 0x40383d35:0x3fcf5f00 0x40383945:0x3fcf6030 0x403786ed:0x3fcf6050 0x40383d65:0x3fcf6070 0x420c02d9:0x3fcf6090 0x420c0925:0x3fcf60b0 0x4202c13e:0x3fcf60d0 0x4202c150:0x3fcf60f0 0x4202c4d5:0x3fcf6110 0x4202cd6b:0x3fcf6140 0x4202cd83:0x3fcf6160 0x4202cda1:0x3fcf6180 0x42023545:0x3fcf61a0 0x42016d9d:0x3fcf6230 0x4201b892:0x3fcf62f0

  #0  0x40377eb6:0x3fcf5ec0 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
  #1  0x4037dd39:0x3fcf5ee0 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
  #2  0x40383d35:0x3fcf5f00 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x40383945:0x3fcf6030 in multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:259 (discriminator 1)
  #4  0x403786ed:0x3fcf6050 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:382
  #5  0x40383d65:0x3fcf6070 in free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
  #6  0x420c02d9:0x3fcf6090 in operator delete(void*) at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
  #7  0x420c0925:0x3fcf60b0 in operator delete[](void*) at /builds/idf/crosstool-NG/.build/xtensa-esp32s3-elf/src/gcc/libstdc++-v3/libsupc++/del_opv.cc:35
  #8  0x4202c13e:0x3fcf60d0 in AsyncWebSocketMessageBuffer::~AsyncWebSocketMessageBuffer() at .pio/libdeps/ESP32_S3_16_8_YELLOW_AP/ESP Async WebServer/src/AsyncWebSocket.cpp:203 (discriminator 1)
  #9  0x4202c150:0x3fcf60f0 in std::_Function_handler<void (AsyncWebSocketMessageBuffer* const&), AsyncWebSocket::AsyncWebSocket(String const&)::{lambda(AsyncWebSocketMessageBuffer*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebSocketMessageBuffer* const&) at .pio/libdeps/ESP32_S3_16_8_YELLOW_AP/ESP Async WebServer/src/AsyncWebSocket.cpp:855
      (inlined by) _M_invoke at /home/mroy/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:297
  #10 0x4202c4d5:0x3fcf6110 in std::function<void (AsyncWebSocketMessageBuffer* const&)>::operator()(AsyncWebSocketMessageBuffer* const&) const at /home/mroy/.platformio/packages/toolchain-xtensa-esp32s3/xtensa-esp32s3-elf/include/c++/8.4.0/bits/std_function.h:687
      (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::remove(AsyncWebSocketMessageBuffer* const&) at .pio/libdeps/ESP32_S3_16_8_YELLOW_AP/ESP Async WebServer/src/StringArray.h:128
      (inlined by) AsyncWebSocket::_cleanBuffers() at .pio/libdeps/ESP32_S3_16_8_YELLOW_AP/ESP Async WebServer/src/AsyncWebSocket.cpp:1226
  #11 0x4202cd6b:0x3fcf6140 in AsyncWebSocket::textAll(AsyncWebSocketMessageBuffer*) at .pio/libdeps/ESP32_S3_16_8_YELLOW_AP/ESP Async WebServer/src/AsyncWebSocket.cpp:957
  #12 0x4202cd83:0x3fcf6160 in AsyncWebSocket::textAll(char const*, unsigned int) at .pio/libdeps/ESP32_S3_16_8_YELLOW_AP/ESP Async WebServer/src/AsyncWebSocket.cpp:963
  #13 0x4202cda1:0x3fcf6180 in AsyncWebSocket::textAll(String const&) at .pio/libdeps/ESP32_S3_16_8_YELLOW_AP/ESP Async WebServer/src/AsyncWebSocket.cpp:1104
  #14 0x42023545:0x3fcf61a0 in wsSendTaginfo(unsigned char const*, unsigned char) at src/web.cpp:151
  #15 0x42016d9d:0x3fcf6230 in processDataReq(espAvailDataReq*, bool, IPAddress) at src/newproto.cpp:549
  #16 0x4201b892:0x3fcf62f0 in rxCmdProcessor(void*) at src/serialap.cpp:362
```

nlimper, 2024-01-21:
Try this one! https://github.com/jjwbruijn/OpenEPaperLink/commit/83bd0ec17721528cddc7d4e44da8888095462bfd

mooseroy, 2024-01-21:
Thanks. Pulled and built that commit and uploaded to AP. Will monitor it over the next day to see how it goes and let you know.

mooseroy, 2024-01-22:
So far this is working much better.  I'll let it run the rest of the day and close the issue if nothing panics.  Thanks for all the help.

closed 2024-01-22 by mooseroy