OpenEPaperLink/OpenEPaperLink #207: PCB for the S3_C6_NanoAP running an ESP32-S3 and ESP32-C6 Feature request: Providing the KiCAD files

TimSchimansky, 2024-01-24:
**Is your feature request related to a problem? Please describe.**
A Gerber file is provided (as well as list of parts in the readme)

**Describe the solution you'd like**
Providing the KiCAD files to open the possibility to order a PCB from e.g. JLCPCB with component assembly.

**Describe alternatives you've considered**
Reverse engineering from the images, but its unclear, which components go where (the capacitors are not marked)

ele0n, 2024-01-24:
I would also be interested in KICAD files or other working files with placement.

atc1441, 2024-02-04:
See attachment, without guarantees :)
[pap.csv](https://github.com/jjwbruijn/OpenEPaperLink/files/14157086/pap.csv)
[bom.csv](https://github.com/jjwbruijn/OpenEPaperLink/files/14157087/bom.csv)

closed 2024-02-04 by atc1441

TimSchimansky, 2024-02-04:
I will try this, thank you!